GrapheneOS/Camera #185: Missing advanced modes: Pixel 6 Pro

lbdroid, 2022-04-17:
According to source code ( https://github.com/GrapheneOS/Camera/blob/main/app/src/main/java/app/grapheneos/camera/CamConfig.kt#L1449 ) several advanced modes like PORTRAIT, NIGHT, HDR should be available in this camera software, however the only modes available are QR, CAMERA, VIDEO.

This is on a Pixel 6 Pro running GrapheneOS, which surely should support all modes.

Zanthed, 2022-04-17:
These rely on CameraX vendor extensions which have not been released yet. The code detects if the extensions are available and uses them if available. If there is no vendor extension for it, it's not used.

closed 2022-04-17 by Zanthed

lbdroid, 2022-04-17:
So to clarify, these modes are implemented in preparation for something that does not exist?

thestinger, 2022-04-17:
It exists. It isn't available on Pixels yet. It works fine on the Samsung devices where it's already available.

lbdroid, 2022-04-18:
Huh, that seems counterintuitive.
Thanks.

alexmo1997, 2022-04-27:
Just out of curiosity, is it know if google is even planning to release vendor extensions for pixel devices or anything regarding that? It's weird that it's their library and samsung released vendor extensions while google did not.

thestinger, 2022-04-27:
CameraX vendor extensions will be available for Pixels. It's in active development. Need to be patient and wait for it to happen. It's not in our control.

There's no overlap between the CameraX team and the Pixel hardware team. The people working on AOSP where Camera2 extensions were added are also a separate group. Pixels are simply another OEM from the perspective of those people and they can't make it happen faster.