GrapheneOS/Camera #189: Pixel 6 Pro: Does not zoom out all the way

lbdroid, 2022-04-20:
Up until now I was under the impression that GrapheneOS camera was able to "zoom out" to the same level as Google camera, since GrapheneOS camera could switch to the wide angle sensor reading the same 0.7x on the back side of the phone, and to 0.9x on the front side of the phone.

However because I need use of HDR (note 1) for protographing real estate this weekend, I decided to give Gcam-services-provider (https://github.com/lukaspieper/Gcam-Services-Provider) and Google Cam (via Aurora Store) a try in order to achieve that.

Turns out that on both sides of the phone, Google camera is able to zoom out substantially more than GrapheneOS camera.

Here are some examples. Note that the phone is held on a table in precisely the same location for both programs.

Google camera fully zoomed out, BACK side:

[image]

GrapheneOS camera fully zoomed out, BACK side:

[image]

Google camera fully zoomed out, FRONT side:

[image]

GrapheneOS camera fully zoomed out, FRONT side:

[image]

note 1: GrapheneOS camera does not support HDR yet since it would be provided through an extension that is not available on Pixels, "yet" (hopefully will be eventually).

ASerbinski, 2022-05-10:
@thestinger -- you commented on an upstream bug for this issue here; https://issuetracker.google.com/issues/230013960?pli=1

Looks like upstream has identified the root cause of the bug and has offered a solution, which is to disable video stabilization mode. I've implemented the relevant change in Nextcloud Talk, which corrects the issue: https://github.com/nextcloud/talk-android/pull/2016

I think it would be nice to have a toggle to control this mode.

Zanthed, 2022-05-10:
I don't think we want to use Camera2 APIs. Camera2 interop _might_ be okay, but any Camera2 APIs beyond that we don't want to use and strictly use CameraX and let upstream deal with it.

thestinger, 2022-05-10:
Using this part of the Camera2 API via Camera2 interop is fine.

ASerbinski, 2022-05-10:
PR: https://github.com/GrapheneOS/Camera/pull/211