GrapheneOS/Camera #201: Consider adding "filterTouchesWhenObscured"

francescoassereto, 2022-04-30:
I tested the latest version of the Secure Camera Application (Version 29) on [AppSweep](https://appsweep.guardsquare.com). Of course, I don't know how reliable is this type of analysis. However, it appears that there are some Elements of `CaptureActivity` and `MoreSettings` which  are not protected against tapjacking. The solution here would be add `filterTouchesWhenObscured="true"` to the relevant view elements.

Analysis: https://appsweep.guardsquare.com/builds/1687e1e1-74a7-48f8-a249-84355e170175?filter=%7B%22Severity%22%3A%22medium%22%7D&page=1

closed 2022-06-02 by MHShetty

MHShetty, 2022-06-02:
Hi @mielediedera,

Tap-jacking in case of our camera app isn't a major security issue as such, although we have added the `filterTouchesWhenObscured` attribute to certain (root) views of the activities of our application.

Thanks a lot for performing that test on our app and reporting this issue @mielediedera!

thestinger, 2022-06-02:
I don't really think we should do this. It's not the approach taken by the rest of the OS. It's used extremely sparingly elsewhere.

thestinger, 2022-06-02:
I don't think it makes sense to address something like this on a case-by-case basis. It's a systemic problem requiring OS solutions. There's already https://developer.android.com/about/versions/12/behavior-changes-all#untrusted-touch-events in Android 12. If we're going to start using this property manually we would need to decide exactly why and where we're going to use it. I don't want to use it globally in the Camera app alone. It just doesn't make sense.

commit c26b68aa (2022-06-02): Prevent tap-jacking in certain UIs (wherever required)
(Fixes #201)