GrapheneOS/Camera #202: Pixel XL Display Ratio Issue

IzumiSenaSora, 2022-04-30:
When i open app 
There's a little bar in top of this app 4:3 or 16:9 ratio same small bar

[image]

[image]

While switching from camera to video or qr scan you will see more black border all side of the screen! 

[image]

And if i pull notification panel once that top bar become bigger 

[image]

and while that top bar is bigger if i switch from camera to video or qr code side border gets more larger... 

[image]

qr code mode get cut icons on top while first opening

[image]

after pulling notification icons in qr code get down....sorry i am bad at English!... 

[image]

Device Pixel XL

r3g-5z, 2023-02-23:
Is this still an issue?

MHShetty, 2024-10-26:
Hi @IzumiSenaSora,

We're currently primarily working on migrating our camera app to Jetpack Compose and would attempt to try and fix this issue in the process.

We'll update this thread once there is any progress or changes to test specifically for Pixel XL regarding this issue.

Thanks a lot for following back on this issue. We'll get back with some update soon!