GrapheneOS/Camera #204: QR Scan freezes camera

mikitakahazekura, 2022-05-02:
https://user-images.githubusercontent.com/24222032/166289408-6572094c-a4bd-44e9-b8fe-06dc5df0ad45.mp4
Steps to reproduce:
1) Open Camera
2) Switch to QR Scan
3) Camera viewer freezes
Device: OnePlus 6
OS: DivestOS 18.1
Build Number: RQ3A.211001.001
Kernel: 4.9.282-dos.p132-gd57522d38ec1-dirty
#1 Sat Apr 9 14:12:58 EDT 2022
Logcat:
[logcat.txt](https://github.com/GrapheneOS/Camera/files/8604746/logcat.txt)

Zanthed, 2022-05-03:
This appears to be a library bug in Material and not our issue.

Zanthed, 2022-05-04:
@mikitakahazekura Please try with Camera v31.

thestinger, 2022-05-14:
Another user also reported an issue on the OnePlus 6. It seems that this is a OnePlus 6 OS bug that's likely also present on the stock OS.

thestinger, 2022-05-14:
@mikitakahazekura Please try again with version 36 or version 37 to see if the CameraX 1.1.0-rc01 update fixed this. It's likely still an issue but knowing for sure that's it's still an issue will help me report an issue to CameraX.

SkewedZeppelin, 2022-05-14:
I can still reproduce this issue on that device.

thestinger, 2022-05-14:
It's possible that the issue is caused by newer Android versions using Scudo instead of jemalloc for this process. It appears to be a memory corruption bug crashing the camera service which you can see in the log that was posted. It's definitely an OS issue but it's possible we could somehow work around it or CameraX could work around it. It would be extremely helpful to know if this issue occurs on the most recent stock OS release for the OnePlus 6 or 6T since if it doesn't then it's likely a bug which was in the stock OS but got uncovered by later AOSP changes.

If you look at the hardened_malloc repository, you can see the 12.1 branch has a workaround for Pixel 3 / Pixel 3 XL memory corruption bugs in the same generation of Qualcomm Camera service as this device. OnePlus 6 appears to be using a 32-bit process so hardened_malloc isn't relevant and it could be getting uncovered by Scudo instead, although Scudo doesn't uncover the issue on the Pixel 3 that's still present.

SkewedZeppelin, 2022-05-14:
I already apply the hmalloc camera exception to all camera libraries on DivestOS

thestinger, 2022-05-14:
Going to close this since it's definitely an OS camera bug due to the crash and it's fairly likely there's nothing we can do to avoid it without forking CameraX and implementing a nasty workaround for it. The only chance of a workaround on our side would be trying to lower the resolution to 640x480 and I don't think that will successfully work around it.

We currently request 960x960 which usually ends up giving us 1280x960 or 1920x1080 (which is the maximum it will request for image analysis).

It might be possible to reproduce it with a CameraX Image Analysis sample like https://github.com/android/camera-samples/tree/main/CameraXAdvanced.

closed 2022-05-14 by thestinger

SkewedZeppelin, 2022-05-14:
I was able to still reproduce it with scudo disabled for 32-bit.
Still likely not worth the trouble

thestinger, 2022-05-14:
@SkewedZeppelin Can you try that CameraX image analysis sample?

SkewedZeppelin, 2022-05-14:
Works perfectly, and is very fun to point at things!

thestinger, 2022-05-14:
@SkewedZeppelin Could you try deleting the line with `setTargetResolution(Size(960, 960))` in `app/src/main/java/app/grapheneos/camera/CamConfig.kt` to see if QR scanning works with the default 640x480 resolution? I think there's a decent chance it does. If that's not the problem, I'm really not sure what it could be.

thestinger, 2022-05-14:
We're also using auto-focus on the scanning square with a 2 second timer. It's possible that the device has broken region-based auto-focus support. Another thing to try would be in `app/src/main/java/app/grapheneos/camera/ui/activities/MainActivity.kt` deleting the code in `startFocusTimer` so that it doesn't do anything:

```
    fun startFocusTimer() {}
```

SkewedZeppelin, 2022-05-14:
> startFocusTimer

That was it!
QR Scan works perfectly with it disabled.

thestinger, 2022-05-14:
@SkewedZeppelin If you rapidly tap to focus in the main camera mode, does that freeze the camera too? The only difference is that in the QR scan mode we use a region (the whole scanning square) instead of a point. Could be that any region-based auto-focus is broken due to the bug in the camera service and crashes it.

It would be helpful to undo the removal of startFocusTimer and instead try removing this call to it:

https://github.com/GrapheneOS/Camera/blob/main/app/src/main/java/app/grapheneos/camera/ui/activities/MainActivity.kt#L203

That would disable the repeated focus but not the initial focus. Could also try increasing the timer value here:

https://github.com/GrapheneOS/Camera/blob/main/app/src/main/java/app/grapheneos/camera/ui/activities/MainActivity.kt#L1299

It's currently 2000 milliseconds. If it was increased to 5000, maybe it wouldn't crash. It could be that it crashes when trying to focus while a focus while already running if it doesn't know how to cancel it properly, etc. Maybe we could cancel it and then sleep for a few hundred milliseconds before starting the next focus.

thestinger, 2022-05-14:
If the issue ends up being that we use a region, we could add a check for that device model where we use a point instead of a region.

thestinger, 2022-05-14:
Using a point instead of a region is as simple as deleting the `qrOverlay.size` parameter here and only passing the width / height to use the overload for a point instead of a region:

https://github.com/GrapheneOS/Camera/blob/main/app/src/main/java/app/grapheneos/camera/ui/activities/MainActivity.kt#L194-L197

I suspect that might work around the problem so what we need is a check for the OnePlus 6 / 6T device model names there based on android.os.Build.DEVICE / android.os.Build.MODEL which make it use a point instead of a region. It's pretty trivial to do that if that's the issue.

SkewedZeppelin, 2022-05-14:
Disabling the repeat via line 203 or increasing the time still resulted in it locking up.
I made PR as you described.

thestinger, 2022-05-14:
Worth noting that if someone can reproduce this on the stock OS we can report it to CameraX and get them to add a device quirk for the OnePlus 6 and OnePlus 6T disabling focus region support and turning it into a point which will work around this for all apps using CameraX. If it's only happening with OSes updating AOSP beyond what the stock OS used to provide then we probably can't get them to deal with it though.

thestinger, 2022-05-14:
We've had a user on the Play Store confirm this happens with the stock OS so we've filed https://issuetracker.google.com/issues/232569772 in addition to merging the workaround for it. We can remove the workaround once CameraX adds a quirk, ships it in a month release and we update to that.

thestinger, 2022-06-01:
@SkewedZeppelin Can you test version 42 and make sure it still works on the OnePlus 6 or 6T? It turns out the CameraX SurfaceOrientedMeteringPointFactory API has a counterintuitive format for the size parameter and we were passing it the wrong way. SurfaceOrientedMeteringPointFactory normalizes the x and y coordinates based on the width/height you pass to it but does not normalize the size. Means we were passing an overly large size and is likely why we were triggering a memory corruption bug in their camera service.

thestinger, 2022-06-01:
https://github.com/GrapheneOS/Camera/releases/tag/42

SkewedZeppelin, 2022-06-01:
Version 42 works well!