GrapheneOS/Camera #218: Video resolution defaults to 4K

mariobrostech, 2022-05-13:


mariobrostech, 2022-05-14:
After looking into this more, the switch in resolutions doesn’t appear to take effect until the settings panel is opened. When this happens, the camera will adjust its cropping accordingly, indicating the resolution change.

thestinger, 2022-05-14:
Trying to reproduce this but I don't seem to be able to do it.

mariobrostech, 2022-05-14:
I was able to find a way to reliably reproduce this issue on my device. The steps I followed, as well as the result, can be seen in [this screen recording I made](https://we.tl/t-AUITMEW6Cd).

thestinger, 2022-05-14:
@mariobrostech We're able to reproduce it. I'm removing your comment since you might have accidentally included some personal info in the video.

thestinger, 2022-05-14:
The lockscreen settings are currently deliberately separate and are also temporary. The idea behind this was to stop people with access to the locked device being able to mess with your settings persistently.

thestinger, 2022-05-14:
Could it just be a side effect of that?

mariobrostech, 2022-05-14:
Possibly, but I’ve noticed this sometimes bleeding into the regular app’s settings as well (though I can’t find a reliable way to reproduce this in particular). Regardless, if the lock screen is using temporary settings, it should inherit those that are already set by the user (or, it should at least provide an option for this to be the case).

thestinger, 2022-05-14:
We could implement that instead of doing what we are right now.

thestinger, 2022-05-19:
Should be resolved by v38.

closed 2022-05-19 by thestinger