GrapheneOS/Camera #227: Request for acceptance: Publish in F-Droid

TheOneWithTheBraid, 2022-05-19:
Are you as developers okay in case I open a PR containing F-Droid metadata? This is one of the few well-working camera apps for Android, I'd love to include it into F-Droid hence.

Zanthed, 2022-05-19:
We're not adding our apps to F-Droid.

closed 2022-05-19 by Zanthed

thestinger, 2022-05-19:
There are multiple reasons we wouldn't be okay with it.

F-Droid has serious privacy, security and usability issues which for the most part they're uninterested in resolving. We consider it a legacy technology which needs to be replaced. Our Camera app is available in our own app repository which we're developing as a replacement for GrapheneOS users. Our app repository can be used outside GrapheneOS and the client works on other Android 12+ operating systems.

Unofficial builds of our apps need to use their own app id and branding rather than the official app.grapheneos.camera / app.grapheneos.camera.play and the official icon. It's confusing for users and results in serious usability issues. Reuse of app ids for builds signed with different keys is a violation of the way app ids are meant to work. We own the grapheneos.app domain and no one else should be using app.grapheneos as a namespace. We would consider it trademark infringement and our trademarks are being officially registered to make it easier to take legal action than with common law trademarks.

Redistributing the official builds of our apps also doesn't work for us because it results in bad usability for people using both our app repository and the other app repository. It also gives people delayed updates with no guarantee those people keep updating it. They may decided to stop as they have done for other apps and leave it permanently outdated, which isn't okay with us. F-Droid has a poor track record when it comes to keeping apps updated and removing insecure unmaintained apps without even basic security maintenance to keep it targeting a recent API level, updating libraries (often with fixed vulnerabilities) and fixing issues in the app itself often because the new target API level will force dealing with it.

F-Droid project members have been extremely hostile towards GrapheneOS and have engaged in a severe libel and harassment campaign personally targeting me. We want nothing to do with any of them. If any more of our code (including via a hostile fork) from any of our projects is published through F-Droid or any repository run by anyone associated with it, we will be taking legal action based on the documented history of libel. They've crossed serious lines and are well into the territory of making it easy to pursue legal action against them. If they do anything else to antagonize us, that will be happening.

Multiple F-Droid developers work for a hostile organization involved in an organized campaign of misinformation against GrapheneOS and libel targeting our project members.

F-Droid has an irredeemably malicious, underhanded and untrustworthy project team. It doesn't help that the software and the repository they maintain are both very poorly conceived, implemented and maintained. We want nothing to do with F-Droid.

Many good open source apps including Signal are not available in F-Droid. We'll be working on convincing developers to make their non-Play-Store app variant self-updating instead of relying on F-Droid. Ideally, app developers will also get their app removed there to avoid the serious usability issues from F-Droid's app id reuse and delayed updates. From our perspective, people should be working on getting apps out of F-Droid and doing their own self-updating especially with Android 12 supporting unattended updates by an app's installer including apps being able to update themselves automatically after the user explicitly approves an initial self-update.

TheOneWithTheBraid, 2022-05-19:
I understand the concerns. I entirely agree that the architecture of central signing keys is a security vulnerability and agree on that there should be a good future alternative to it. As of now, I unfortunately do not see any in open source world. Let's hope for some future improvement. Thanks a lot for the explanation.

thestinger, 2022-05-19:
The best alternative is for apps to update themselves. Android 12 allows them to do that automatically. Once the user gives the app the permission to install/update apps, it can update itself, and after the first self-update is explicitly proved by the user it can do subsequent updates automatically.

Android itself implements package signature verification and downgrade protection in the OS so apps don't need to reimplement that and just need to download an update apk or set of split apks and verify / install them through the OS API including checking app id and making sure they only update themselves. It's really not a lot of work because they don't need to implement any kind of repository, etc. just a way to fetch latest version and then download the apk(s) to verify/install via OS APIs. They don't actually need to implement metadata signing or anything like that since they don't need a repository listing apps and can just check the app id / version in the apk and use OS verification APIs.

thestinger, 2022-05-19:
> 
I understand the concerns. I entirely agree that the architecture of central signing keys is a security vulnerability and agree on that there should be a good future alternative to it. As of now, I unfortunately do not see any in open source world. Let's hope for some future improvement. Thanks a lot for the explanation.

They do have the option to redistribute developer builds of apps but in the vast majority of cases they aren't doing that and it still causes issues. Having the same app id distributed from multiple places is a usability issue and F-Droid can end up falling behind. It will always be delayed, but often substantially delayed. It's better for people to get the apps from our repository.

https://github.com/GrapheneOS/Apps/releases works fine on other Android 12+ operating systems and will update itself and our Auditor, Camera and PDF Viewer apps. It has proper metadata signing with key rotation support and it has stable/beta/alpha channels where we push out updates via alpha, then beta and then stable similar to what you can do on the Play Store. It has unattended update support even outside GrapheneOS on operating systems where it's not a trusted first party source since Android 12 supports that. If you install an app from there, it can automatically update it going forward. It also has compressed transfers of the apks and we intend to implement full split apk support and delta update support for even more efficiency. There are a lot of advantages to it.

TheOneWithTheBraid, 2022-05-19:
Yes, I entirely agree. Personally, I anyway tend to publish apps in as many stores as possible, including such evil ecosystems as F-Droid but I entirely understand if developers don't like this.

One more quick question: Does the Camera app support self-updating? Or does it rely on updates from an external app store?

thestinger, 2022-05-19:
We have our own app repository and app repository client so our apps delegate updating to that. Look at https://github.com/GrapheneOS/Apps/releases which is usable outside GrapheneOS. It's called Apps and will update itself and our 3 standalone apps. We'll probably **eventually** have Vanadium in there too but there are major blockers to address before we can do that.

thestinger, 2022-05-19:
The way we expect people to use our Camera app outside of GrapheneOS is to either get it from the Play Store (app.grapheneos.camera.play app id) or install Apps (same app.grapheneos.camera release as on GitHub Releases page) and then install it from Apps which will keep itself and Camera updated.

thestinger, 2022-05-19:
Camera supports Android 10+ because Android 10 is the oldest major release of Android with security support.

Apps requires Android 12+ because we didn't specifically try to support outdated Android branches like we did for Camera where we're going out of the way to support Android 10 and 11. We don't currently plan to bother because we expect most people using Auditor / Camera / PDF Viewer are getting them via Play Store and users on an alternate OS without Play are probably going to be on Android 12 even if the vendor didn't provide an Android 12 update.

Officially, each Android major release gets 1 year of monthly (monthly security patches, other bug fixes, other minor improvements) / quarterly updates (major improvements and sometimes even major new APIs like Android 12.1) and then 2 years of LTS support with only security updates once the next major release is out. Sometimes they get up to another year of security-only LTS support.

Since we're privacy and security focused, we only really care about support Android versions with security support, so our Android 10+ focus can become Android 11+ later this year, then Android 12+. We don't plan to **remove** Android 10 support from Camera at that point but we wouldn't bother with Android 10 support in a new app like a standalone gallery app once it was end-of-life.

WhyNotHugo, 2022-06-06:
> F-Droid has serious privacy, security and usability issues

This is the first I hear about this. Do you have a link or reference to more details on the privacy and security issues elsewhere? I'm interested in reading more on this; looks like I've been missing out on something important.

> Many good open source apps including Signal are not available in F-Droid. We'll be working on convincing developers to make their non-Play-Store app variant self-updating instead of relying on F-Droid. 

A nice details of f-droid is that they'll clarify whether an application is open source, or whether it has anti-features, etc. This fulfils the same role as what packagers do on desktop OSs (they kinda _are_ packagers, just for mobile). The idea of downloading random binaries from the upstream developer is weird, because there's no obvious way of knowing if the vendor is reputable or not. I'm not implying F-Droid is _necessary_ for this; I'm implying that they currently fulfil this role and it's unclear what the alternative would be on Android.

Any suggestions on this topic?

> From our perspective, people should be working on getting apps out of F-Droid and doing their own self-updating especially with Android 12 supporting unattended updates by an app's installer [...]

The odd thing about this is that it duplicates the burden of packaging, hosting and implementing updated mechanisms into **every single** application. The burden on developers is non-trivial here.

With regards to this approach: if your vision is that applications should self-update, then why the recommendation to download your [`Apps`](https://github.com/GrapheneOS/Apps) app and install the camera through there? Isn't this the opposite of this self-updating idea?

As an end user, the idea of having a dedicated `Apps` app with the sole purpose of just downloading and updating the camera app seems super inconvenient and honestly, just bloat.

> The way we expect people to use our Camera app outside of GrapheneOS is to either get it from the Play Store (app.grapheneos.camera.play app id) or install Apps (same app.grapheneos.camera release as on GitHub Releases page) and then install it from Apps which will keep itself and Camera updated.

So F-Droid is unacceptable due to privacy concerns, but going through Play Store (which requires signing up with Google) is officially supported... Is using Google considered a privacy-friendly alternative here?

thestinger, 2022-06-06:
> With regards to this approach: if your vision is that applications should self-update, then why the recommendation to download your [Apps](https://github.com/GrapheneOS/Apps) app and install the camera through there? Isn't this the opposite of this self-updating idea?

Apps is self-updating and handles updates for our other apps. If we only had a single app it wouldn't make any sense to have Apps. It's in no way the opposite of self-updating. Apps is our own update system for our own apps.

WhyNotHugo, 2022-06-06:
Right, for users who have multiple Graphene apps it's role is maybe clearer... But for someone just using the `app.grapheneos.camera` app it's weird.

Following the same mindset, any developer with more than one app would ship their `Apps` app to update all their apps -- I think you can see the problem I'm aiming to describe, right?

thestinger, 2022-06-06:
> This is the first I hear about this. Do you have a link or reference to more details on the privacy and security issues elsewhere? I'm interested in reading more on this; looks like I've been missing out on something important.

F-Droid violates the app source security model by mixing repositories together and bypassing user / device policy for trusted sources (privileged extension makes this much worse, and doesn't make sense rather than using the modern APIs with unattended update support), uses a legacy metadata signing approach without complete file verification or any support for key rotation, builds their apps on a poorly secured server running Debian stretch (not even stable or oldstable which have serious issues with missing security fixes) and in general has a bunch of serious privacy and security issues with their app. Main members of the project are involved in covering up vulnerabilities / security weaknesses on multiple occasions and can't be trusted. They've also organized a raid on one of our issue trackers and otherwise acted maliciously towards us.

F-Droid also incorrectly reuses app ids for their own builds with their own signatures which is a violation of the app id system and causes serious usability issues. Users who install an F-Droid build in one profile get an error if they try to install the official release in another. Every build variant of an app is supposed to have a unique app id, especially incompatible ones with incompatible signatures.

> So F-Droid is unacceptable due to privacy concerns, but going through Play Store (which requires signing up with Google) is officially supported... Is using Google's services considered a privacy-friendly alternative here?

Play Store follows the app source security model and developers are in full control of what they publish and which app id used for it. Play Store enforces a minimum target API level for new apps and app updates which enforces many enforced privacy and security restrictions on apps not enforced by F-Droid in any way.

> A nice details of f-droid is that they'll clarify whether an application is open source, or whether it has anti-features, etc. This fulfils the same role as what packagers do on desktop OSs (they [...]

An app being open source doesn't mean it's more private, secure or trustworthy. F-Droid is not doing any significant auditing / review of the apps and does not enforce any serious standards. Their "anti features" are mostly misguided and they mark them very incompletely and inconsistently. If anything you're just proving that they give people a false sense of privacy/security while most of the apps in their repo are poorly maintained / unmaintained and have known security vulnerabilities due to lack of library updates. They also often delay updates that are available for weeks, months or indefinitely. Play Store review approval at least usually takes only ~2-4 days and that review is not something F-Droid does for app updates (they just update them and run automated checks).

thestinger, 2022-06-06:
> Following the same mindset, any developer with more than one app would ship their Apps app to update all their apps -- I think you can see the problem I'm aiming to describe, right?

That isn't what was stated, and no, we do not see the problem.

thestinger, 2022-06-06:
Most of this was already stated above in https://github.com/GrapheneOS/Camera/issues/227#issuecomment-1132203158. I'm unsure why you quoted the first few words while ignoring the vast majority of what was written.

WhyNotHugo, 2022-06-06:
> F-Droid violates the app source security model by mixing repositories together and bypassing user / device policy for trusted sources (privileged extension makes this much worse, and doesn't make [...]

Wow, those actually do sound like very problematic issues. Thanks for taking the time to explain all that.

I've encountered glitches due to app id mismatches -- now I think I understand better why that is.

WhyNotHugo, 2022-06-06:
> An app being open source doesn't mean it's more private, secure or trustworthy.

I agree, of course not. Though the opposite _is_ true; if an app is entirely closed, there's no way any proper audit can be done, so (whenever viable) it's best to just rule them out fast and early.

>>Following the same mindset, any developer with more than one app would ship their Apps app to update all their apps -- I think you can see the problem I'm aiming to describe, right?
>
> That isn't what was stated, and no, we do not see the problem.

Well, as Graphene developers, it seems very convenient for you to make your own `Apps` app to auto-update your apps. But as downstream users, it just becomes "the app to update the graphene camera app".

If every developer (or team of developers) followed the same model, then for the 20 applications on my phone, I'd have 20 accompanying updater application for it -- instead of each app updating itself individually.

Again, for each developer the model is convenient, but the sum of it does not scale well for end users.

thestinger, 2022-06-06:
Another example of how they screw up usability with profiles is that they publish an outdated version of their own app on their site through the main download link. That download link is shown on every app page including the F-Droid app page since it's there to encourage you to install via F-Droid itself instead of downloading manually. When users try to install F-Droid in a secondary user, the version is older than what they probably have in existing profiles already and they get an error. This becomes a support burden for GrapheneOS. F-Droid has refused to fix this for years. They implied in a conversation that they see it as a sensible punishment for users on operating systems not bundling F-Droid and expected us to bundle their app rather than fixing it by keeping their link up-to-date.

They also tried to excuse it by saying the latest version may not work properly and yet F-Droid will try to update you to that version, including doing it automatically if you use their privileged extension. It makes no sense as an excuse.

Our app repository has arbitrary release channels for each of the apps in the repository, similar to the Google Play Store. We use alpha and beta channels in addition to stable. Like the Play Store, these are a way to release production releases of the exact same variant of the app which is implied by app id. We push updates to alpha, then beta, then stable in order to reduce the issues which reach stable due to our organized testing coordinated in #testing:grapheneos.org on Matrix. F-Droid is missing this kind of thing, and that's a major issue too. Since they are building other people's apps they also don't know when to move them to the release channels, but the fact that the updates are extremely delayed means it usually doesn't matter anyway... unless they caused an issue with their undocumented changes they make to some of the apps.

thestinger, 2022-06-06:
> Well, as Graphene developers, it seems very convenient for you to make your own Apps app to auto-update your apps. But as downstream users, it just becomes "the app to update the graphene camera app".

When those users open the app they'll also see we have Auditor and PDF Viewer apps. In the future, they'll also see hardened GrapheneOS builds of many third party apps.

I think many app developers would be more than happy to have an installer app where they get to list all their apps since it's essentially a form of non-intrusive promotion for the others apps. If they just have one app or their apps are not meant to target the same group of people like ours then they could just do self-updating in the app itself. For example, WhatsApp could have a proper self-update system for releases outside the Play Store. Android 12 provided unattended update support. Unfortunately many apps are several years behind on adopting modern Android platform features, partly because they don't use AndroidX or other modern development practices properly.

> If every developer (or team of developers) followed the same model, then for the 20 applications on my phone, I'd have 20 accompanying updater application for it -- instead of each app updating itself individually.

The original purpose of Apps was distributing our mirrors of the core Google Play apps for GrapheneOS users to obtain them for using sandboxed Google Play. We are planning to provide a repository with builds of other apps. Other developers are not planning to do either of these things. We never proposed that other developers make an app repository client like ours. We suggested that apps should perform proper self-updating in a modern / secure way since it avoids involving a third party.

thestinger, 2022-06-06:
It's quite likely F-Droid will be severely compromised due to their bad security practices. Openly admitting to still using Debian stretch on multiple servers is the tip of the iceberg. A lot of their infrastructure is very badly written and maintained. We're trying to prepare our community for what is very likely to happen. When that happens, the vast majority of apps users have installed from F-Droid's repository can have a compromised update pushed out. Any new apps people install from it can be compromised. It's a massive single point of failure and is an unnecessary third party introduced as an additional trusted party. The user has to trust the app developer and the OS. They do not have to trust a third party app store like F-Droid. F-Droid is neither the OS provider or the app developer. It's a mistake to think they are performing some kind of meaningful / useful review which would catch a malicious update... It's extremely unlikely they would. At best the fact that updates are so delayed would often insulate users from that kind of event, if F-Droid actually reacted in time and blocked the update...

thestinger, 2022-06-06:
When this happens, we want to have minimized the % of GrapheneOS users that are impacted and we also want to minimize the impact it will have on the reputation of GrapheneOS. If most people using GrapheneOS get their apps from the developers and the Play Store via either sandboxed Google Play or Aurora Store, it will greatly reduce the impact. From our perspective, based on what we have seen, this is an event we need to prepare for and at the moment we are unfortunately not well prepared because a substantial portion of our community heavily uses F-Droid. Multiple people in our community are working towards providing better alternatives, which we can promote as we've done with Molly (hardened Signal) and Bromite.

TheOneWithTheBraid, 2022-06-07:
@thestinger Are you familiar with [SkyDroid](https://skydroid.app/)? (It has to do nothing with F-Droid.) I think, this might be a solution preserving the freedom.

SkyDroid is a decentralized app store for Android ~~allowing~~ *enforcing* to use own signatures. Unlike F-Droid, it's up to the developers to provide instant app updates, distribution is done decentrally by Sia Skynet and verified by DNSSEC.

Could this maybe be a solution for users who do not want to use Graphene OS but still want to profit of the great applications you develop?

thestinger, 2022-06-07:
People can use our app repository client outside GrapheneOS on Android 12 and later:

https://github.com/GrapheneOS/Apps/releases

TheOneWithTheBraid, 2022-06-07:
Yes, thanks a lot. I was more likely asking whether you could additionally support distribution via SkyDroid. As @WhyNotHugo said, for most non-Graphene users, Graphene OS Apps would be an application to install a single application, while with SkyDroid, there is an alternative possibly fulfilling the security requirements on a 3rd party app distribution place.

Would you consider this? Is it worth opening an extra issue on this or wouldn't you like to support this?

thestinger, 2022-06-07:
If we publish the app with the same app id in 2 places and a user uses that SkyDroid on GrapheneOS, it will try to update our apps asking users to accept it but we want Apps updating them.

thestinger, 2022-06-07:
Also, if they bridge to F-Droid as that page states, they inherit all the issue from F-Droid for that including app id conflicts.

locness3, 2023-01-21:
Thank you for the detailed criticism of F-Droid. Do you have anything to link to regarding the said harassment from the F-Droid community towards GrapheneOS maintainers?

thestinger, 2023-01-21:
https://twitter.com/DanielMicay/status/1547286521597894657 is one of many examples.

thestinger, 2023-01-21:
The main F-Droid developer has been involved in spreading libel about me in public chat rooms and forums for years along with engaging in bullying and harassment. There is ample evidence of that in multiple places. I've documented some of it on my feed. What I linked is a different one of them doing it and then that  main developer showing up to try to hide what happened, push fake stories about it and mislead people as part of their dishonest character assassination campaign. There is hard proof through the archive provided there compared with the thread after they deleted all this and lied about what happened, along with the archive of their chat room showing coordinated talking points and harassment. This is one of many of these incidents. They have continued doing this including recently. Should also be noted they are paid by organizations they pretend not to be associated with and push what's in the interest of those organizations while making explicit statements which present them as an unrelated group aligned with them in their attacks on me. Look at the thread on their forum where Hans is spreading blatant libel about me and fabrications about GrapheneOS. Did you know he was involved in the takeover attempt on GrapheneOS in 2014 and took money from Copperhead? He still continues the attacks with their same debunked talking points today. It's right there in their forum.