GrapheneOS/Camera #231: Actions debug APK

xslvrxslwt, 2022-05-20:
Maybe add a downloadable debug apk to GitHub [actions](https://github.com/GrapheneOS/Camera/actions)?

Zanthed, 2022-05-20:
I don't really see why that's necessary.

closed 2022-05-20 by Zanthed

Zanthed, 2022-05-20:
We'd rather not have people run random debug APKs and report false bug reports/issues and make things confusing for alpha/beta testing.

xslvrxslwt, 2022-05-20:
> I don't really see why that's necessary.

Already running actions, it's already there, just export the apk so we can use it before the actual stable release 😅

thestinger, 2022-05-20:
It's not meant for people to use and we don't want issue reports from people testing the development branch.

xslvrxslwt, 2022-05-21:
> It's not meant for people to use and we don't want issue reports from people testing the development branch.

Oh, okay, sorry