GrapheneOS/Camera #235: EXIF data randomly removed even when told not to

Saroumane, 2022-05-24:
Google Pixel 3a
Camera : Latest : 39, Installed : 39
"remove EXIF data after capture" : turned off since the  6th of March.

As noted here ( https://github.com/GrapheneOS/os-issue-tracker/issues/1036#issuecomment-1104016857 ) : 
on the 20th of April I exported the pictures from my phone : 
_I took 362 pictures. (I always invoke Camera the same way, by double pressing the top right physical button)
Among them 11 pictures (taken on 4 different days) have had their EXIF data completely removed._

Today (24th of May) I did another export of new pictures from my phone. 
I took 351 pictures. Among them 44 pictures (taken on 8 different days) have had their EXIF data completely removed.
As a workaround, I have to retag them manually. (partially, because it takes too much time)

I don't see a pattern, but here is the filenames of the pictures without EXIF data.

```
IMG_20220424_132344.jpg
IMG_20220424_132348.jpg
IMG_20220427_105656.jpg
IMG_20220428_210922.jpg
IMG_20220504_130340.jpg
IMG_20220504_130342.jpg
IMG_20220504_130344.jpg
IMG_20220504_130352.jpg
IMG_20220504_130353.jpg
IMG_20220504_130404.jpg
IMG_20220504_130406.jpg
IMG_20220504_130408.jpg
IMG_20220505_124109.jpg
IMG_20220505_124110.jpg
IMG_20220505_124115.jpg
IMG_20220505_124117.jpg
IMG_20220505_200729.jpg
IMG_20220505_200733.jpg
IMG_20220505_200735.jpg
IMG_20220505_200739.jpg
IMG_20220510_202037.jpg
IMG_20220519_222257_960.jpg
IMG_20220519_222259_793.jpg
IMG_20220519_222301_537.jpg
IMG_20220519_222831_904.jpg
IMG_20220519_222834_118.jpg
IMG_20220519_222835_377.jpg
IMG_20220519_222837_279.jpg
IMG_20220519_222838_377.jpg
IMG_20220519_222839_234.jpg
IMG_20220519_222840_087.jpg
IMG_20220519_222840_767.jpg
IMG_20220519_222841_596.jpg
IMG_20220519_222844_055.jpg
IMG_20220519_222845_130.jpg
IMG_20220519_222846_553.jpg
IMG_20220519_222847_735.jpg
IMG_20220519_222848_921.jpg
IMG_20220519_222851_279.jpg
IMG_20220519_222853_322.jpg
IMG_20220520_113739_939.jpg
IMG_20220520_183046_581.jpg
IMG_20220520_183106_246.jpg
IMG_20220520_183108_326.jpg
```

thestinger, 2022-05-25:
This is likely because the secure (lockscreen) mode didn't inherit the settings you've configured for the normal mode before version 38.

closed 2022-05-25 by thestinger

Saroumane, 2022-05-25:
Thanks for the explanation.

Saroumane, 2023-07-26:
Sorry to come back here, but it seems there is a regression.
I never touched this setting again ("remove Exif data" set to 'off') and suddenly I have dozens of pictures without Exif data, that I have to timestamp manually. In fact : it's all of them since the  7th of July. I went to camera settings, and the setting was back to 'on'.
I guess a Camera update around the 7th July has changed my settings.
I just want to know if it can happen anytime Camera is updated, and there is no "warranty" that my settings are kept ?