GrapheneOS/Camera #238: Viewfinder does not match recorded video with EIS enabled

enchainingrealm, 2022-05-26:
I am running Camera version 39 on a Pixel 2. To reproduce this bug:

- Enable EIS in video mode.
- Zoom in to the highest setting (7.0x).
- Start a video recording.
- Slowly pan the phone until the desired subject is in the center of the screen.
- Record for as long as desired.

In the recorded video, the subject is at the edge of the screen instead of the center. I except the viewfinder should always match the recorded video to prevent misleading the user while shooting.

thestinger, 2022-05-26:
This is probably just how EIS works on that device.

enchainingrealm, 2022-05-26:
Do you mean EIS is implemented via an Android API that GrapheneOS Camera calls? And thus it's out of scope for GrapheneOS Camera to fix this issue?

thestinger, 2022-05-26:
We use the Camera2 EIS toggle. It's the standard Pixel implementation of EIS. You should try using Google Camera to see if the same behavior occurs. It's possible Google Camera uses a more advanced implementation though.

enchainingrealm, 2022-05-26:
I do indeed get the same behaviour with Google Camera, although it's less severe because the maximum zoom is 5.0x in Google Camera.

thestinger, 2022-05-26:
Is the maximum zoom only 5x when EIS is enabled in Google Camera? It's possible they added an artificial limit to work around this issue, but that limit isn't implemented in the Camera2 API. I don't think we're going to add a workaround specifically for the Pixel 2.

enchainingrealm, 2022-05-26:
In Google Camera, the maximum zoom for photos is 7.0x and the maximum zoom for videos is 5.0x, regardless of whether EIS is enabled.

thestinger, 2022-05-26:
That seems like an artificial limitation they implemented in Google Camera since the Camera2 API still says the maximum zoom is 7x when video is added as a use case. I don't think we're doing anything wrong and it works.

closed 2022-05-26 by enchainingrealm