GrapheneOS/Camera #239: Pictures with low light and flash are black

Yourproblem, 2022-05-26:
I installed camera yesterday. When I want to take a picture with low light the camera flashes but the picture is black. When I focus in dark, there is also no light for focus.
Device is a3y17lte.

thestinger, 2022-05-26:
Does it work properly if you use the torch feature instead of the flash? It sounds like the device has a broken implementation of flash for the Camera2 API and doesn't keep it on long enough. These issues need to be worked around in CameraX which it has done for a few devices already. We can't do much ourselves but we can report it to them.

debie-A320FL, 2022-05-27:
It works properly when we use the torch feature.

thestinger, 2022-05-27:
Google Camera doesn't support that doesn't so you're using some unofficial variant of it. If someone specifically ported it to that device, they must have taken care of what was required.

You could try using Latency mode instead of the Quality mode but it's probably not the problem.

There's nothing we can do about this within our app. These workarounds have to be done by CameraX. It has workarounds for devices with this kind of flash bug already but the list of devices where it's used is far from complete. This has to be reported at https://issuetracker.google.com/issues?q=status:open%20componentid:618491 after reproducing it with one of their sample apps. If it only happens in Quality mode where CameraX waits for focus/exposure lock, that needs to be included in the issue report.

thestinger, 2022-05-27:
This device probably needs to be added here:

https://android.googlesource.com/platform/frameworks/support/+/refs/heads/androidx-main/camera/camera-camera2/src/main/java/androidx/camera/camera2/internal/compat/quirk/FlashTooSlowQuirk.java?autodive=0%2F%2F%2F%2F%2F%2F%2F

Other apps may already hard-wire a workaround for this device.

closed 2022-05-27 by thestinger

farfromrefug, 2022-06-07:
@thestinger thanks for all the details! i reproduced the issue with the sample apps and reported the issue here https://issuetracker.google.com/issues/235172945