GrapheneOS/Camera #244: add support for manually switching camera ids, primarily for devices without automatic Camera2 zoom-based camera switching like Pixel phones

thestinger, 2022-05-28:


thestinger, 2022-05-28:
This is somewhat blocked on CameraX adding APIs but may be possible via the Camera2Interop API already.

ASerbinski, 2022-06-03:
Along related lines, would it be possible to manually select cameras on Pixel instead of relying exclusively on the level of zoom? The problem on Pixels is that you can't, right now, pick a specific camera without having some level of image scaling kicking in and obviously reducing the quality of the shot.

Coming from a pixel 6 pro, the scenarios where image scaling is presumably not present are;

1) First time starting up the main side camera where it comes in at 1:1.
2) Zooming out FULLY on the main side camera where it switches to the wide angle lens without applying image scaling (zoom).
3) Zooming out FULLY on the screen side camera (the default mode on this side is scaled).

There is currently no way to select the telephoto lens without image scaling since it is impossible to select the exact threshold for when it switches.

adam-718, 2022-06-04:
If possible, just add a button to select the ultra-wide lens, just like in the OpenCamera.
Thanks

thestinger, 2022-06-04:
@ASerbinski That might be possible but I expect it would be significantly different.

Pixel 6 Pro telephoto lens switches at a dynamic point rather than at exactly 4x even though the telephoto camera is 4x since there are some disadvantages to using it. For example, it doesn't work as well in low light, so it will extend the zoom or avoid using it at all in low light. Not clear exactly what heuristics they use for this. There may be a way to switch manually with the multi-camera API but I think they present the cameras as a single camera ID with multi sub cameras so handling this would be different from other devices even if it's possible to change this.