GrapheneOS/Camera #246: Documentation for installing this app

staticdev, 2022-05-29:
I have checked out this app and I think it is great. Unfortunatelly, I have Android phones without GrapheneOS and I would very much like to install it.

Could you please add to README.md instructions to install it on normal Android phones?

thestinger, 2022-05-29:
There's an explanation in the release notes:

https://github.com/GrapheneOS/Camera/releases

It can be installed from either the Play Store or our app repository. Our app repository client is available for Android 12+. We expect most people outside GrapheneOS to obtain it from the Play Store.

closed 2022-05-29 by thestinger

staticdev, 2022-05-29:
@thestinger thanks for your response. Unfortunatelly, I still don't have access to Android 12. Also the link for Play Store does not work for me on Android 11. Does this app have a minimum requirement for Android 12?

Zanthed, 2022-05-29:
Our Camera app works on Android 10 and above. What device are you on?

staticdev, 2022-05-29:
Fairphone 4.

thestinger, 2022-05-29:
Camera app supports Android 10 and later. Play Store release should work fine for you.

Our app repository client requires Android 12 or later.

staticdev, 2022-05-30:
This is strange, I also tried with Aurora store, same result. It doesn't find the app by the link on release notes. Does it work for you?

thestinger, 2022-05-30:
Yes, it's working fine. Which device model is it?

thestinger, 2022-05-30:
I don't know why it wouldn't be working. Fairphone 4 is in the device catalog. Play Store says there are 21 users on it.