GrapheneOS/Camera #247: Documentation - Android 12 Regressions

locuturus, 2022-05-31:
Hello, I'm on Android 12 and 13 (non-graphene) devices and you can no longer set a 3rd party camera app as default for the system's camera launch gesture. 

This is a major Android regression in my opinion but beyond how I feel about it what it also means is I cannot access this app's secure mode. Anyone installing this from the play store might want to know it cannot be used on the lock screen, cannot be launched with a double press of the power button, and any mention of secure mode is only for Android 10, 11, and certain ROMs (such as yours).

Note the what's new section refers to features a lot of your potential users can't utilize.
[image]

locuturus, 2022-06-04:
> You can still choose the app launched with the gesture including for the lockscreen on Android 12. Your OEM may have removed support for the feature but it's still there in AOSP and the stock Pixel OS. There's a UX bug on Android 12 which you can hit having multiple camera apps installed though.

I'm using a Pixel 4a on the Android 13 beta, reading Reddit complain about the missing ability to assign the gesture on their Pixels on Android 12, and I've tested a Samsung S20 FE 5G on Android 12. So I'm encouraged, I guess, that you think this is still supported by Google but I'm also deeply skeptical from my experience.

thestinger, 2022-06-04:
It's still fully supported and works on AOSP / stock Pixel OS / GrapheneOS. If you have multiple camera apps installed, it triggers a bug for the lockscreen activity because it tries to ask which one it should use and it doesn't work correctly. There is a way to configure it where it works even with multiple camera apps but we don't have an exact set of steps. You can try disabling / uninstalling the other camera apps and you'll see it works perfectly fine at least on a Pixel. We know this for a fact. It has nothing to do with faith. We're telling you what we can clearly replicate ourselves over and over again.

thestinger, 2022-06-04:
Disable Google Camera. GrapheneOS Camera will be able to launch on the lockscreen. This isn't a reasonable solution because Google Camera is the only system camera app on the stock Pixel OS. If you disable all system camera apps, then there is nothing provider the system media capture intents for other apps to use the system camera, since they don't work with non-system ones since Android 10 (not related to lockscreen issue at all).

It's possible to configure the default camera app in a way that it will work on the lockscreen with multiple camera apps installed. It requires doing things in a very specific way in a specific sequence. You can reset the app associations for each camera app and try to work out the sequence required to get this set up properly. We don't have a guide since we don't actually know what the sequence is to work around it. It is definitely possible and we've confirmed it multiple times.

locuturus, 2022-06-04:
I'll fiddle with it again. Perhaps I'll uninstall all camera apps next. I already tried disabling the system camera and I did also reset all camera app associations already. The double press power button gesture does absolutely nothing at all with Google Camera disabled. Nothing I've tried has allowed me to prompt the open with dialog for cameras whether the phone is locked or unlocked. I also haven't encountered the lock screen bug you mention, because I no longer get that far in the process.