GrapheneOS/Camera #248: Swipe gesture in landscape mode

dhishkyaon, 2022-06-01:
In Google Camera -

https://user-images.githubusercontent.com/88529066/171336137-83a1d787-2598-4d1f-8b9e-9b086e912a20.mp4

In GrapheneOS Camera -

https://user-images.githubusercontent.com/88529066/171336193-181209da-8fd0-4e4a-8213-f7f5436c89c5.mp4

thestinger, 2022-06-01:
You need to describe what you want to see.

dhishkyaon, 2022-06-01:
swipe gesture from Left to Right or Right to Left like Google's Camera app

closed 2022-06-02 by MHShetty

MHShetty, 2022-06-02:
Hi @dhishkyaon,

This issue has now been fixed in the latest changes pushed the repository. The app/fix can be tested either by cloning the repo/pulling the latest changes and then building the app. Please do let us know if any further changes are required by any chance.

Thanks a lot for carefully testing the app and reporting this issue @dhishkyaon!

commit 76dfd31b (2022-06-02): Fix settings dialog transition for landscape mode (Fixes #248)