GrapheneOS/Camera #256: Entire camera module become unresponsive

pixincreate, 2022-06-10:
As said, latest version is rendering the camera module useless unless the device is rebooted.

This happens only upon clicking on the video tab, which overrides the entire camera module resulting in crashing of camera in other apps.

(ZSL turned on)

Device: Nokia 6.1 Plus (DRG)
Android version: 12.1 (LineageOS 19.1)
Architecture: arm64

Zanthed, 2022-06-10:
We will need logs then. This sounds like an OS or hardware bug with the camera that CameraX is going to need to workaround, but we'll need logs.

Zanthed, 2022-06-10:
Please get logs for us if you can.

thestinger, 2022-06-10:
This is a bug in your OS. It's not the fault of our app or CameraX and there's little we can do about it.

thestinger, 2022-06-10:
An app crashing an OS component is always an OS bug. The app can sometimes work around the OS bug but in this case that's done but not enabling ZSL by default. We do enable EIS by default which is known to break the camera services on some poorly implemented devices. Your OS should be disabling support for ZSL and also EIS if the implementation of the features is broken.

closed 2022-06-10 by thestinger

thestinger, 2022-06-10:
You need to report this issue to your OS provider. It's an OS bug.

pixincreate, 2022-06-10:
> You need to report this issue to your OS provider. It's an OS bug.

Reported. 

> You could also try disabling EIS. It may be the combination of switching from using ZSL to using EIS that breaks

I tried that now, no ZSL issue, it doesn't work even EIS is disabled.

The LineageOS developer has acknowledged to check the issue on Telegram.

Zanthed, 2022-06-10:
These logs are weirdly formatted and don't indicate any issues with our app. This is a bug with your OS or hardware.

pixincreate, 2022-06-11:
> These logs are weirdly formatted and don't indicate any issues with our app. This is a bug with your OS or hardware.

I do think so, the issue has been reported to the dev and eventually, an issue will be raised on gitlab

pixincreate, 2022-06-11:
https://review.lineageos.org/c/LineageOS/android_device_nokia_sdm660-common/+/332596

The dev disabled ZSL entirely. But the same works well on GCam.

thestinger, 2022-06-11:
That should resolve the issue since it won't be available but their commit very is very wrong. CameraX only uses Camera1 when it isn't available. It uses Camera2 whenever it's available and uses it in a quite modern way.  Camera1 ZSL isn't even the same API that CameraX is using via Camera2.

> The dev disabled ZSL entirely. But the same works well on GCam.

Google Camera doesn't support your device, so you're talking about a hacked together unofficial port. It likely doesn't use ZSL.

pixincreate, 2022-06-11:
CameraX only uses Camera1 when it isn't available. It uses Camera2 whenever it's available and uses it in a quite modern way. Camera1 ZSL isn't even the same API that CameraX is using via Camera2.

Does this mean that the device itself doesn't support? 

> Google Camera doesn't support your device, so you're talking about a hacked together unofficial port. It likely doesn't use ZSL.

Sad but glad to know about this. I've been GCam mods for all these years.

thestinger, 2022-06-11:
> Does this mean that the device itself doesn't support?

I don't think the ZSL toggle could cause a crash if Camera1 wasn't supported because I doubt they use ZSL for the Camera1 API. I think the device maintainer copied a commit from elsewhere for disabling ZSL without adjusting the message. You can see from https://review.lineageos.org/c/LineageOS/android_device_oneplus_msm8998-common/+/311484 that this commit came from elsewhere. However, I think that's not the original source of the commit and the message was wrong much earlier. People are copying it around either without understanding the issue or without bothering to adjust the commit message. I think the issue is simply that you device has a broken ZSL implementation, and this commit appears to fully disable ZSL, so it won't be an issue anymore despite the reasoning given in the commit message about ZSL being broken with the front camera with Camera1 being totally wrong.

pixincreate, 2022-06-11:
Thank you for the clarification.

pixincreate, 2022-06-13:
Ok, wait, @thestinger, I observed something that I found ridiculous.

The thing is, the entire camera module becomes unresponsive only when ZSL is turned on **and** `OPTIMIZE FOR LATENCY` is selected! If the same is set to `OPTIMIZE FOR QUALITY` it doesn't crash the camera module!

thestinger, 2022-06-13:
No, that's not the problem. ZSL only has an impact in Latency mode. It's ignored in Quality mode.

thestinger, 2022-06-13:
ZSL is only used in Camera mode when Optimize for Latency is enabled and flash is disabled. It isn't used for capturing photos in video mode (even in latency mode) or for the extension modes. It seems like your device crashes when switching from image capture with ZSL enabled to video capture without it enabled since it isn't used for video capture.