GrapheneOS/Camera #257: Recording video on LinageOS gauguin not working

foss-, 2022-06-11:
LineageOS 19.1 2022-06-06, gauguin
Secure Camera v43 or 42

Not sure if Secure Camera or LinageOS updates broke video recording. What happens is:

- Start Video
- image on screen instantly gets stuck while recoding time continues ticking
- as soon as the video is stopped message `Recording too short to be saved` is shown
- video is never saved

Here is [Logcat](https://bin.disroot.org/?4b43d358f1945d3b#BX6dQjVpcFXeosx51e49SdFVGGzLB5miki6G4zeiniru) (30 days), maybe it is useful to help understand what is going on.

thestinger, 2022-06-11:
EIS is used by default when available now. When it's available, there's a toggle for disabling it in the settings dialog for video mode. Please check for that toggle and try disabling it. Your OS may have a broken implementation of EIS.

foss-, 2022-06-11:
Thanks for the speedy response. Disabled EIS in settings but the problematic behavior persists.

Zanthed, 2022-06-11:
Please post logs that aren't restricted behind client-side decryption and WASM.

foss-, 2022-06-11:
Can you be more specific as to what to change?

I stopped Orbot but am still seeing `06-11 17:11:51.002  1322  1885 W ActivityManager: Foreground service started from background can not have location/camera/microphone access: service org.torproject.android/.service.OrbotService` in the log.

MHShetty, 2022-06-11:
Hi @foss-,

> I stopped Orbot but am still seeing `06-11 17:11:51.002 1322 1885 W ActivityManager: Foreground service started from background can not have **location/camera/microphone** access: service org.torproject.android/.service.OrbotService` in the log.

The logs related to Orbot are not related to our camera application. Could be an issue related to Orbot/OS or maybe the service hasn't actually been properly turned off as expected.

Please share the logs in it's decrypted form or if you're concerned about any sensitive data getting via Matrix (@wowowowowowowowowow:matrix.org) to enable us to further help with this issue.

Thanks for trying out our camera application @foss-!

MHShetty, 2022-06-11:
Hi @foss-,

> I stopped Orbot but am still seeing `06-11 17:11:51.002 1322 1885 W ActivityManager: Foreground service started from background can not have **location/camera/microphone** access: service org.torproject.android/.service.OrbotService` in the log.

The logs related to Orbot are not related to our camera application. Could be an issue related to Orbot/OS or maybe the service hasn't actually been properly turned off as expected.

Please share the logs in it's decrypted form or if you're concerned about any sensitive data getting via Matrix to enable us to further help with this issue.

Thanks for trying out our camera application @foss-!

thestinger, 2022-06-11:
It has nothing to do with Orbot.

foss-, 2022-06-11:
Log is freely accessible, no encryption or required password for access.

Additional testing revealed the issue also happens with v41 which iirc worked fine previously. This would suggest the issue was not introduced by a recent Secure Camera update but instead lies with LineageOS itself.

Will bring this up with the maintainer of gauguin. Sorry for not testing this thoroughly enough before reporting. If help is needed, will re-open.

Thanks for helping me out so kindly.

closed 2022-06-11 by foss-