GrapheneOS/Camera #258: Add toggle for date EXIF metadata

repunset274, 2022-06-14:
EXIF removal is an important and appreciated feature in Graphene's camera. However, some users may still like to include the date a photo was taken for organizational purposes (while removing other potentially fingerprinting-related exif). Would the devs be open to adding a toggle to preserve the "Date taken" field, while still stripping the remaining metadata?

thestinger, 2022-06-14:
The timestamps are really the main thing we're removing. The rest of it is mostly info about the camera / device model / exposure settings. The date is still available in the name of the photos.

thestinger, 2022-06-14:
We could implement this but it's not really something we feel is that useful and isn't going to be a very high priority.

repunset274, 2022-06-24:
Thank you for the consideration!

repunset274, 2022-06-24:
I envision this could be handled a couple of ways; either 3-4 toggles asking which data to remove (camera model, Date Taken, etc etc); or the inverse, of which to keep. The current "remove exif after capture" would likely be toggled on by default (as it is now) - but toggling this off could allow for the other toggles to appear, (or go from grey to usable) to either remove (or maintain) subsections of attributes.

Donkey-Doug, 2022-09-05:
Would also love this feature