GrapheneOS/Camera #259: Integrated video player pauses music even when video has no audio

catdevnull, 2022-07-07:
When recording a video without audio and then playing it back with the built-in player, it will unnecessarily pause music.

MHShetty, 2022-07-07:
Hi,

I'm really sorry but I was really unable to get you. Like are you referring to any music/audio that's being played in background that gets paused while you watch a video that doesn't have any audio in our in-app gallery?

Also, thanks a lot for trying out our camera app @catdevnull!

catdevnull, 2022-07-07:
Yes, that's exactly what I mean. Thank you for developing the app, it's the best OSS camera experience I've ever had, by far.

MHShetty, 2022-07-08:
Hi @catdevnull,

It's our pleasure. Glad that you liked our app :)

Checking whether the video has audio by its metadata using the `MediaRetriever` class and then explicitly setting the audio to false via it's controller might help resolve this as an issue. My personal system that I was working on earlier is currently broken and is under repair, I'll try working on this issue once my system gets fixed.

Thanks for filing this issue @catdevnull!

PR #302 (2022-10-12): don't request audio focus for muted video
fixes #259
closed 2022-10-12 by thestinger