GrapheneOS/Camera #262: exif gps location not saved

lapoigne, 2022-07-07:
Google Pixel 6
Build number : SQ3A.220605.009.B1.2022063000
Camera : version 43
remove EXIF data after capture : off 
location : enabled

I have enabled camera location and when I go to the photo detail I can't see the gps location on exif tags
I can see the app requesting location to the OS.

I do some tests with OpenCamera app:
 - If I use original camera API, location is saved on exif
 - If I use Camera2 API, location is not saved, like GrapheneOS camera app

lapoigne, 2022-07-08:
Just got an update to version SQ3A.220705.003.A1.2022070600 and location is now saved

closed 2022-07-08 by lapoigne

trenfoight, 2023-02-27:
Im using pixel 7 pro and having same issue

thestinger, 2023-02-27:
Have you enabled location and granted access to location? Test with GPSTest to see if you're getting a GNSS (satellite-based) location lock. If you're on GrapheneOS, there's only GNSS-based location for the OS location API, not network-based location, so it needs some time to get a lock.

trenfoight, 2023-03-06:
GPS is on...im using phonetrack with my nextcloud server with no problem