GrapheneOS/Camera #264: Unable to fully launch since v43

juliangaal, 2022-07-08:
I am unable to fully launch the app since v43.

The app itself opens, but the viewfinder never initializes with the camera feed. Reboot doesn't fix.

OS: GrapheneOS
Device: Pixel 3

Any additional info I can provide you with?

thestinger, 2022-07-08:
Your camera hardware isn't working anymore. We can't fix that here.

closed 2022-07-08 by thestinger

juliangaal, 2022-07-08:
Oh really?! Damn, time for an upgrade :D Thanks for your work, Daniel. Cheers.

thestinger, 2022-07-08:
Pixel 3 has been end-of-life since after October 2021 so it's already time for an upgrade. Get a Pixel 6, Pixel 6 Pro or the upcoming budget Pixel 6a because they have a minimum of 5 years of support from launch. Previous devices had a minimum of 3. We continued providing 3rd gen Pixel extended support releases but they don't provide full security updates since most of the device-specific updates haven't been available since after their EOL in October 2021.

juliangaal, 2022-07-08:
Got it.