GrapheneOS/Camera #265: Ability for manually changing white balance

rightsaidfred99, 2022-07-10:
Hi, 

It would just be good to have some advanced settings or a slider to change white balance. 
Camera does a good job but sometimes it will get white balance wrong in situations like sunsets.

Xonar92, 2022-07-23:
Since I’ve started to use the app heavily for photos and videos I have also noticed some random problems with auto white balance.

As a result I get this overly blue tint in my shots from time to time.

thestinger, 2022-07-23:
White balance is provided by your phone, not this camera app. It only uses what's provided to it. We could offer more configuration in the future but there aren't CameraX APIs for it yet. It may be possible to use Camera2Interop as we did for EIS support.

Zanthed, 2022-07-24:
Google Camera is not a CameraX app so it's not relevant here.

MHShetty, 2024-06-12:
We'll most likely need to implement this using Camera2Interop API. A feature request was filed by us earlier regarding this on CameraX's issue tracker, but it doesn't seem to have been officially made into a CameraX feature yet (Link: https://issuetracker.google.com/issues/208583141)

empratyush, 2024-06-12:
@MHShetty previous work on [dev](https://github.com/empratyush/Camera/tree/dev-wb-13-10-23) branch might be helpful.

MHShetty, 2024-06-12:
Hey @empratyush!

Ohh you seem to have already implemented this on one your branches. Thanks a lot for the valuable work and mentioning this out!

Are you looking forward to perform a PR for this feature? Is there any work that's pending before the feature is complete that I could help out with?

empratyush, 2024-06-13:
> Is there any work that's pending before the feature is complete that I could help out with?

I am not sure. iirc ui is not that good.