GrapheneOS/Camera #270: Import QR scanned data

inpharmaticist, 2022-07-29:
I think it would be convenient for the camera app QR scanner to have the option to detect/import certain formats. For example, a QR formatted as .vcf could show the option to `add to contacts`.

jfrederickson, 2023-02-28:
Yeah - related, I just tried scanning a QR code for a wifi network, and was hoping for this to happen. (Most phones these days prompt you to join the network when you scan a QR code like that.)

thestinger, 2023-02-28:
That feature is supported through the Wi-Fi add network QR scanner.

jfrederickson, 2023-02-28:
Ah, that's good to know. Would still be convenient if the camera's QR code scanner could detect special formats like this though and send you to the right place automatically.

empratyush, 2023-10-18:
see #389

closed 2023-10-18 by empratyush