GrapheneOS/Camera #281: Add a video mode where the screen stays black and records in ultrawide.

troublemaker101, 2022-09-12:
The black screen won't be recognizable because of oled screen that most pixels have. Real life use case : Police violence, undercover journalists etc.. I don't have any idea how to implement it. Cheers.

thestinger, 2022-09-12:
I don't really see how that could be stealthy and probably belongs in a dedicated app.

closed 2022-09-12 by thestinger

thestinger, 2022-09-12:
Pointing a phone at something is what makes it not stealthy. I don't think having the screen black will make much difference.

troublemaker101, 2022-09-12:
Please, but really in situations with the police it can really be helpful. A german court just odered it to be illegal to film police investigation even if you wan't to secure evidence. A phone screen either taking a video or being 100% black thanks to oled would be a big difference.