GrapheneOS/Camera #285: Night Sight Not Showing

mercertom, 2022-09-16:
Pixel 4a 5g
Checked settings, nothing

thestinger, 2022-09-16:
Settings -> Display -> Night Light (in Color section).

thestinger, 2022-09-16:
I misinterpreted this as a GrapheneOS issue report about something else. Will provide a proper response.

thestinger, 2022-09-16:
Our Camera app supports all of the CameraX vendor extension modes: HDR, Night, Portrait (Bokeh), Face Retouch and Auto. Pixels don't provide CameraX vendor extensions yet so it's not available. All 5 modes are available on Samsung phones from the past couple years. This is just something Pixels need to implement. These features currently have their processing implemented in Google Camera, other than baseline HDR+ for image capture and HDRnet for previews (and also videos on 6th gen Pixels) which are available in our Camera app already in the normal modes.

closed 2022-09-16 by thestinger

mercertom, 2022-09-16:
2 tweets affirming vendorX stuff including Night Sight would be added to Graphene Camera. What happened that it's "not planned"?

mercertom, 2022-09-16:
So no Night mode on Graphene Camera on Pixels until vendorX extensions get provided by Google?

thestinger, 2022-09-17:
GrapheneOS Camera already has working CameraX vendor extension support. On a Samsung phone from the past couple years, it has all 5 vendor extension modes already. It isn't something that needs to be implemented for our camera app. You're wrong that it's missing from our app. It's not provided by Pixels yet. It's provided by our app on devices supporting it.

mercertom, 2022-09-17:
At no point did I say it's missing from "our" app. I said it's "not showing" in the issue title, and asked, "so no Night mode on Graphene Camera on Pixels until vendorX extensions get provided by Google?"