GrapheneOS/Camera #286: Camera Access During Screenlock Issues

Occupant9589Carpet, 2022-09-19:
One thing I have run across while testing how to record video in a situation where I don't want it known because I am trying to collect evidence, is that it is not possible to record video with the screen locked. Also if someone was able to take my phone away from me they would be able to delete any video or pictures that were taken without having to unlock the phone to do so.

I hope that in a future update this could be addressed. It would be nice to double click the power button, hit record, hit the power button once to go to locked, then when the power button is touched again the pin would have to be entered before being able to get back to the camera app. As well as still recording while the screen is locked. Thank you.

thestinger, 2023-02-21:
Recording video was implemented a while ago.

You can use the back gesture to close the activity so that there's a new secure lockscreen session without the ability to view or delete the pictures from previous ones.

thestinger, 2023-02-21:
Recording video was implemented a while ago.

You can use the power button or back gesture to close the activity so that there's a new secure lockscreen session without the ability to view or delete the pictures from previous ones.

thestinger, 2023-02-21:
Recording video was implemented a while ago.

You can use the back gesture to close the activity so that there's a new secure lockscreen session without the ability to view or delete the pictures from previous ones.

https://github.com/GrapheneOS/Camera/issues/208 covers closing it automatically when the screen is turned off.

closed 2023-02-21 by thestinger

Occupant9589Carpet, 2023-03-19:
Sweet. Thanks for the tip. I didn't try the back button since I always have subconciously locked with the power button for years whenever locking any of the phone models I have ever owned. I have never thought to use the back button.