GrapheneOS/Camera #308: Panorama mode

Sesam31, 2022-10-17:
Hi, the app is really nice but the panoramic mode would really give it an edge and make it super useful!

thai-taco, 2022-11-17:
I would also like to express my deep desire for this functionality. For travelers, hikers, and outdoors people especially, but really for near everyone, panorama mode is a core feature of a camera, not having it is a blocker that prevents switching in full

AdamLuchjenbroers, 2024-02-26:
Would also like to request this functionalitym

Jeje292, 2024-05-10:
+1 to add this feature 
I searched it everywhere in app's parameters, without success 😵‍💫

Aary-Thatte, 2026-09-05:
Yes

bigtechstruggler, 2026-10-23:
Hi GrapheneOS Camera maintainers,

I wanted to share **Fold Camera v143.3.3**, an independent, unofficial fork of GrapheneOS Camera.

Since this issue concerns panorama support, I thought the implementation might be of interest.

The project includes:

- 180° and 360° panorama capture with on-device image stitching
- Motion tracking and image alignment
- Memory management and image-quality safeguards
- Portrait photography with on-device processing
- Dual Photo and Dual Video support
- Additional camera controls and foldable-device improvements

**Complete source code:**
https://github.com/bigtechstruggler/Fold-Camera

**APK release:**
https://github.com/bigtechstruggler/Fold-Camera/releases/tag/V143.3.3

The complete source code is now publicly available, including the build configuration and required model files.

I understand that upstream integration would require independent code review, compatibility testing and possible architectural changes.

I'm sharing this in case any of the implementations or ideas could be useful to the GrapheneOS Camera project.

Feedback is very welcome.

Thank you for your work on GrapheneOS Camera!

Best regards,
bigtechstruggler

RankoR, 2026-10-23:
Hey @bigtechstruggler, could you please submit a PR?

bigtechstruggler, 2026-10-23:
Hi @RankoR,

Thanks for your interest! I've submitted the pull request as requested:

https://github.com/GrapheneOS/Camera/pull/788

It includes the complete Fold Camera v143.3.3 feature set, not just panorama support. This includes 180°/360° panoramas, portrait processing, Dual Photo/Video, customizable camera modes with drag-and-drop reordering, the More menu, and foldable support.

The latest GitHub Actions builds are passing.

I'd really appreciate your feedback. Thanks for taking a look!