GrapheneOS/Camera #310: Can't set it as default camera.

KuroRagnar, 2022-10-25:
I tried both apps, playstore and github release but it only list the stock camera.

Poco M3
MIUI global 12.5.9 (RJFMIXM)
Android 11

[image]
[image]

thestinger, 2022-10-25:
If you install a different 3rd party camera app such as Open Camera, does it get listed there?

thestinger, 2022-10-26:
I think this is a bug in the OS and there's not much we can do about it. Would appreciate if you updated your review with the additional information since it doesn't seem it's an issue with our app.

closed 2022-10-26 by thestinger