GrapheneOS/Camera #318: Black icons in Whatsapp file explorer (joining a picture)

Azertixe, 2022-11-02:
Hello. When i go to send a picture in Whatsapp, i see black icons instead of thumbnails with pictures made with GrapheneOS Secure Camera.
➡️ It only happens when i choose SD card for default storage.
➡️ Icons back to normal after a restart. 
➡️ It's just a visual issue. I can join and open this pictures normally.
Any solution ?

Samsung Galaxy A3 2017
LineageOS 17.1
[image]

thestinger, 2022-11-02:
This appears to be a bug in your OS. The media store isn't indexing files saved by SAF. It's possible we could set them to be indexed manually.

Azertixe, 2022-11-02:
Hi. Thanks for your answer.
I haven't the bug using this camera :
https://www.celsoazevedo.com/files/android/google-camera/dev-greatness/f/dl9/
But I have the bug with Open Camera for example.

closed 2023-09-29 by empratyush