GrapheneOS/Camera #319: Phone capable of 4K@30 &1080p@60FPS records 4K@25 & 720p@30FPS

Rexadev, 2022-11-03:
4K@30fps, 1080p@30/60fps - https://www.gsmarena.com/xiaomi_redmi_5_plus_(redmi_note_5)-8959.php#:~:text=4K%4030fps%2C%201080p%4030/60fps
https://www.qualcomm.com/products/application/smartphones/snapdragon-6-series-mobile-platforms/snapdragon-625-mobile-platform#Overview:~:text=Video%20Capture%20(30%20FPS)%3A%204K%20Ultra%20HD%20video%20capture

thestinger, 2022-11-03:
This is what your phone implements via the Camera2 API, which is abnormal, but it's common for certain brands not to provide a good Camera2 implementation. It's possible that the phone makes the other frame rate modes available but CameraX doesn't offer frame rate configuration yet. There's already an issue filed about adding frame rate configuration once CameraX supports it. It's likely your phone just doesn't make the modes available via Camera2.

closed 2022-11-03 by thestinger

Rexadev, 2022-11-04:
@thestinger Add option to set it in the settings

thestinger, 2022-11-04:
That's planned, see https://github.com/GrapheneOS/Camera/issues/51. It can't be done yet since CameraX doesn't support it.