GrapheneOS/Camera #327: Portrait-oriented photos show up as landscape in certain apps

AnonRS3, 2022-11-24:
It appears that the app saves orientation information to EXIF data rather than actually taking the photo at the correct orientation. This leads to apps which do not save or read EXIF data to display portrait-oriented photos as landscape.

thestinger, 2022-11-24:
It's not possible to take the photos in a rotated orientation. It's not how the image sensor works. Using EXIF orientation metadata is the normal approach and those apps are broken if they don't handle EXIF orientation metadata which is part of basic support for displaying an image. Rotating an image with less than 100% JPEG quality would be lossy due to how the compression works. The default is 95% JPEG quality in Latency mode.

thestinger, 2022-11-24:
Report bugs to any app displaying images incorrectly. It would happen with other camera apps too unless they post-process images to rotate them, which as I said is usually going to be lossy.

closed 2022-11-24 by thestinger