GrapheneOS/Camera #330: UI Changes

harishkumarankr2004, 2022-12-03:
The Secure Camera App has a lot of vertical space above the navigation bar which is simply been wasted.

It also has a lot of space wasted above the mode switcher.

Allocating the shutter button and co a bit lower will allow for better reach for the end user.

The zoom slider must be placed in a vertical way for easy accessibility because it will be used very often.

The zoom slider could now be placed above the mode switcher due to the extra gaps we have created by pushing the shutter and co down.

Switching between Video and QR code requires sliding twice in order to reach it. The sliding speed of the mode switcher should be increased along with haptics.

The bottom arrow icon is extremely small and it overlaps with the time shown in notification bar.

Change the orientation of icons based on the auto rotate option present in android.

[image]
[image]

thestinger, 2024-01-16:
It does change when you start recording video. It goes from a circle to a square (stop button). We could make this more obvious.

szmarczak, 2024-09-30:
I also see the issue where the notification bar obstructs the `arrow` settings button.