GrapheneOS/Camera #342: Bug: Video skips first seconds

firefoxlover, 2023-02-09:
I was able to reproduce this twice.

I record a video normally and save it. 

Result:
First few milliseconds, maybe 0,25s show, then a big timeframe like 3s-15s are skipped and the video goes on normally until the end.

Pixel 4a, up to date apps.

MHShetty, 2024-06-10:
Hi @firefoxlover,

This ideally shouldn't be happening. Could you please try reproducing the issue again while trying to record the logs during this time? This could possibly help us better understand where the issue currently lies.

Thanks for trying out our app and reporting this issue!

MHShetty, 2024-08-05:
Hi @secretmango,

Could you please try testing our latest release for this issue?

Thanks a lot for your time and help!

secretmango, 2024-08-19:
Seems to be fixed!

closed 2024-08-19 by secretmango

MHShetty, 2024-08-19:
Glad to know that @secretmango

No issues at all, thanks a lot for reporting this issue and helping out with testing!