GrapheneOS/Camera #344: Add toggle for Mirror Selfie

aeoliam, 2023-02-21:


thestinger, 2023-02-21:
There's already a toggle for this in more settings.

closed 2023-02-21 by thestinger

aeoliam, 2023-02-21:
i couldn't find it

[image]

thestinger, 2023-02-21:
Save image as previewed. We could make the wording clearer in the future.

aeoliam, 2023-02-21:
Ah I see. Thanks for your help!

aeoliam, 2023-02-23:
The saved picture is not mirrored even when the "Save image as previewed" option was on.

[image]
[image]
[image]

aeoliam, 2023-02-23:
The picture's saved is not mirrored in Gallery and other apps.

1. GrapheneOS Camera
[image]
[image]
2. QP Gallery
[image]
2. MiXplorer Image Viewer
[image]

thestinger, 2023-02-23:
EXIF orientation metadata isn't universally fully supported. This isn't an issue with our Camera app and it's not going to be implemented by post-processing the image and losing a lot of latency (it would also have to use 100% JPEG quality for the actual image capture then lower it to the configured value).