GrapheneOS/Camera #349: Autofocus seems disabled since last update (60)

raphaelyancey, 2023-03-15:
I'm not sure this is linked but it seems pretty close. Anybody else experiencing this?

Tried to reboot with no effect. I could downgrade to confirm but I'm not sure how?

Thanks.

thestinger, 2023-03-15:
The main change was updating to https://developer.android.com/jetpack/androidx/releases/camera#1.3.0-alpha04.

raphaelyancey, 2023-03-15:
> Which device model?

Google Pixel 4a

> The main change was updating to https://developer.android.com/jetpack/androidx/releases/camera#1.3.0-alpha04.

Yes I'm sure this wasn't introduced intentionally :)

Just tried with Google Camera app and still no autofocus. I'm not quite familiar with how it works, is it a physical focus? What's the next step I should take to debug this?

thestinger, 2023-03-15:
It doesn't sound like there's an issue with the camera app if it doesn't work in other apps. Autofocus is working fine on Pixel 4a with latest Beta release of GrapheneOS here. This sounds like a hardware failure now.

closed 2023-03-15 by thestinger