GrapheneOS/Camera #350: Add a zoom out camera feature like from the Gcam

MarkCyberSecDev93, 2023-03-21:
[image]

thestinger, 2023-03-21:
How should we decide which zoom values to show? They show minimum zoom, 1x zoom and the base value where the telephoto lens can be used which we can't easily determine across devices. Also, their 4x zoom button on Pixel 6 Pro and Pixel 7 Pro often does not actually trigger switch to telephoto camera despite that being a common misconception because the zoom value it switches is dynamic and often is around 4.7x or 4.8x for typical lightning indoors, on a cloudy day or in the evening.

thestinger, 2023-03-21:
Duplicate of https://github.com/GrapheneOS/Camera/issues/87.

closed 2023-03-21 by thestinger

MarkCyberSecDev93, 2023-03-21:
Ok, I'll delete this issue as it has its own duplicates