GrapheneOS/Camera #356: Publish on IzzyOnDroid

marek22k, 2023-04-05:
Hello,

I really like the GrapheneOS camera app, but I don't use GrapheneOS myself. So I have to manually check GitHub periodically to see if there is a new update. As I saw in the issue tracker, the app is not supposed to be released on F-Droid. Is it possible to release the app to an alternative repo like IzzyOnDroid? For example, IzzyOnDroid fetches the release from GitHub when a new tag is created, scans it via VirusTotal and then delivers it via F-Droid.

r3g-5z, 2023-04-05:
Duplicate of  https://github.com/GrapheneOS/Camera/issues/227

We're not adding our apps to F-Droid and we request that no one publishes or delivers our apps anywhere on places like F-Droid, including IzzyOnDroid.

> So I have to manually check GitHub periodically to see if there is a new update.

You can install our Apps app if you're on Android 12 or above. It works fine there to receive Camera, PDF Viewer, Apps, etc updates.
https://github.com/GrapheneOS/Apps/releases/tag/17

> scans it via VirusTotal

Not something we want to do or integrate. Enumerating badness by scanning an APK on AV engines is not what this project is about. You can do that yourself.

closed 2023-04-05 by r3g-5z

marek22k, 2023-04-05:
That's a pity to hear.I have discovered Obtainium (an app) as an alternative, which can automatically pull updates from GitHub - not optimal, but it works.

For those who have the same problem as me: As an alternative to Graphene's Apps I have discovered Obtainium (an app) as an alternative, which can automatically pull updates from GitHub - not optimal, but it works.

thestinger, 2023-04-05:
Why don't you just use our app repository client? It provides fully automatic updates even outside GrapheneOS.

thestinger, 2023-04-05:
It wouldn't make much sense for each of our apps to include code for updating themselves from our app repository. There are multiple apps in the repository. Vanadium is currently GrapheneOS specific but it will eventually be available elsewhere.

There's an explanation of why we had to make our own app repository system instead of using F-Droid at https://gist.githubusercontent.com/thestinger/d96e591c87463784d4f80f98fbf02a34/raw/d0c3ae842c778cbebf37b60da81ce4c9fe842829/gistfile1.txt. It has fully automatic updates even in operating systems not integrating it, unlike F-Droid, which is a significant advantage for users outside GrapheneOS.