GrapheneOS/Camera #370: Changing screen orientation in gallery resets current image to first one

mb720, 2023-07-15:
Hi and thanks a lot for GrapheneOS!

When I'm in the image gallery and rotate the screen, e.g. from portrait to landscape, the gallery resets to show the first image in the gallery.

Then I have to swipe back to get back to the image I wanted to see.

I'm using a Pixel 6 Pro.

The camera app is:

>version 63
>app.grapheneos.camera
>versionCode 63
>targetSdk 33
>minSdk 29

I tried working around this by changing the gallery app to some different app, but according to the discussion in [this feature request](https://github.com/GrapheneOS/Camera/issues/111), that's not possible.

empratyush, 2023-09-29:
>When I'm in the image gallery and rotate the screen, e.g. from portrait to landscape, the gallery resets to show the first image in the gallery.

Unable to reproduce this. Is this about built-in gallery app in GrapheneOs?

mb720, 2023-09-29:
The bug still happens on my device.

>Is this about built-in gallery app in GrapheneOs?

Yes, the one that opens when you click here inside the camera app:

[image]

The camera app is:

    version 63
    app.grapheneos.camera
    versionCode 63
    targetSdk 33
    minSdk 29

tsxv478, 2024-03-07:
Can confirm that this happens when swiping between full screen images in both grid and filmstrip views, except with grid view it's not the first image in the gallery, but rather the one you last opened. To reproduce: open any image from the grid view, swipe to another one without exiting full screen view and then rotate the screen.

mb720, 2024-03-13:
I'm not sure if this was fixed in version 66, which I'm using now, but the issue doesn't occur at the moment. Thanks for fixing this!

@tsxv478: Note that the issue I referred to happened in the camera app's gallery, not the app named "Gallery" that supports filmstrip view and grid view.

MHShetty, 2024-09-06:
The issue has been fixed in the PR mentioned above #440

closed 2024-09-06 by MHShetty

mb720, 2024-09-06:
I'm using the camera app version 74 and this seems to be fixed. Thanks a lot 👍