GrapheneOS/Camera #371: Geotagging does not work correctly

acinonyx, 2023-07-30:
```
app.grapheneos.camera
versionCode 63
targetSdk 33
minSdk 29

Geotagging enabled
EXIF stripping disabled

```

The GPS tags are added to the EXIF metadata but the values are empty. I have tried various combinations of permissions to rule out a regression of https://github.com/GrapheneOS/Camera/issues/149.

dreamsyntax, 2023-09-08:
```
pixel 6 on GOS TQ3A.230805.001.2023090600

app.grapheneos.camera
versionCode 63

targetSdk 33
minSdk 29
```

Cannot reproduce when using while unlocked.

If location toggled on after a reboot with no location query the camera app has a warning that the geotag is skipped as notification not available. Next photo successfully tags as the location query completes.

empratyush, 2023-09-25:
I am closing this issue as the camera app doesn't seem to be the issue.

closed 2023-09-25 by empratyush

ProgrammerWhoPrograms, 2025-03-31:
> I am closing this issue as the camera app doesn't seem to be the issue.

Then what is the issue?

I am having this issue too and it is causing problems for me when adding images to [Vespucci](https://vespucci.io/)

```
Pixel 6a (bluejay)

GOS build number: 2025032500

Camera app:
app.grapheneos.camera
versionCode 83
targetSdk 35
minSdk 29

Vespucci app:
de.blau.android
versionCode 3101 (21.0.0.0)
targetSdk 34
minSdk 21
```

dreamsyntax, 2025-03-31:
> Then what is the issue?
> 
> I am having this issue too and it is causing problems for me 

Location query must be completed before photos. Open an app that uses hi accuracy location, eg Google Maps and allow location to fully resolve once per reboot before taking photos

peix2, 2026-02-24:
**This issue should be reopened.**

Pixel 9
app.grapheneos.camera
versionCode 90
targetSdk 35
minSdk 29
GOS build number: 2026021201

Location is on.
Camera has permission set for location
There is app in the background for GPS tracking and GPS position is locked and correct.
Camera setting for geotagging is on.
Camera setting to remove EXIF (despite comment it should not delete geotag) is disabled - I can see all other EXIF data.

No matter how many pictures I'm making and how long phone is unlocked there is no geolocation data saved to picture. At the same GPS tracking app is running in background having location established. 
There is clearly something wrong with camera app, either the way it obtains location data or store data to file. 
**Please reopen this issue.**