GrapheneOS/Camera #373: QR scan area shows as black square

nonremittal, 2023-07-31:
Xiaomi Mi A2 running Android 10 and the Camera app v. 63.

The active scanning area shows up as an opaque black box:
[image]

Scanning itself works fine. Haven't experienced this in any other apps with similar designs (e.g. Firefox for Android's QR scanner appears as intended).

Duggi-Niharika-Sai-06, 2023-11-28:
Hello,

I came across this issue labeled as  in your repository, and I'm very interested in contributing to it as part of a school project.

I believe working on this issue will not only help me gain practical experience in, but also contribute positively to your project. I've reviewed the contribution guidelines for your repository and am ready to adhere to them.

Could you please let me know if it's possible for me to work on this issue? If there are any specific requirements or starting points you'd like me to know, I'd be glad to hear them.

Thank you for considering my request. I'm looking forward to the opportunity to contribute to your project.

Best regards,
Neethu Duggi

thestinger, 2023-11-28:
It looks like a bug with the device and it's not clear what can be done about that.

MHShetty, 2024-06-12:
Hi @nonremittal,

Are you currently active to help us out with resolving this issue?

MHShetty, 2024-06-12:
Hi @nonremittal,

Are you currently active and willing to help us out with testing a solution for this issue?

Thanks a lot for reporting this issue and trying out our app!