GrapheneOS/Camera #374: Add support for saving and previewing images in JPEG XL format (with compression options)

life00, 2023-08-20:
I have recently discovered a new image format called JPEG XL. I was fascinated by its capabilities compared to other standard image formats (e.g. PNG, JPEG). The only issue I found was software support. 

It would be nice if the Camera app had an option to save photos as JPEG XL with compression options and ability to preview them.

If someone is concerned about actually viewing `.jxl` images in GrapheneOS (because the default Gallery app doesn't support JPEX XL), then I found an easy solution of using [jxlviewer](https://github.com/oupson/jxlviewer) android app until official support is provided.

It is likely that this image format will become the next big standard so this implementation would not be a waste of time.
For more information check: [official website](https://jpeg.org/jpegxl), [community website](https://jpegxl.info/) (especially recommended), [wikipedia page](https://en.wikipedia.org/wiki/JPEG_XL).

orgua, 2023-12-25:
for more context: this issue has a slight overlap with #14 - raw would be really nice and enable so much more. sad, that cameraX has no priority in this regard. 

also the race between jxl and avif is still ongoing :)

RedNumber1, 2024-04-13:
This app is bare bones. Better to ask devs of open camera or Libre camera

MHShetty, 2024-09-06:
The CameraX team has accepted the issue and we should have something that can be used to implement JPEG XL (or direct support) in one of the upcoming releases

Chealer, 2024-11-29:
Supporting JPEG XL this way may require significant efforts. There is however a request for Android to support JPEG XL. [Open Camera's equivalent ticket](https://sourceforge.net/p/opencamera/tickets/1019/) has some background.

Thanks for filing this @life00 but this is an issue tracker, so entries should be titled as such. I would suggest:

> No support for JPEG XL image format