GrapheneOS/Camera #379: (Feature request) physical camera button support

Turbolqk, 2023-09-12:
Hey guys, great camera app you have there! 
I'd like to ask if it's possible to add physical camera button support, like the ones on Sony devices etc.
I know your OS supports Pixel only, but your apps are used outside GrapheneOS itself too!
Thanks!

empratyush, 2023-09-21:
what device you are on? we can add support for generic shutter(enter) button.

Turbolqk, 2023-09-21:
Xperia Xz1 compact. It has a two-step button - half press for focus and full press to take a pic.

empratyush, 2023-09-21:
I hope this extra camera button is generic button otherwise it would be impossible to utilize this.

Turbolqk, 2023-09-21:
Well both LineageOs' Aperture and Open Camera utilize it, so its probably this.

empratyush, 2023-09-21:
> Well both LineageOs' Aperture and Open Camera utilize it, so its probably this.

unsure about other apps. As i said earlier if it a generic button (which most likely it is) then it should be easy to add support for it.

PR #380 (2023-09-22): add support hardware camera, focus and zoom buttons
closes #379
closed 2023-09-22 by thestinger