GrapheneOS/Camera #381: Unable to save video when locked

hephaisto, 2023-09-30:
When my device is locked, I can double-press the power button to get to the camera and switch to video. I can start and stop video and it is saved properly on the device. However, when I instead start recording the video and press the power button again to switch off the screen, I get "Unable to save this video: Error code 4" when I switch it back on. The complete video is lost, even the part before pressing the power button.

Device: Pixel 4

See also: https://discuss.grapheneos.org/d/7728-unable-to-save-video-when-locked

hephaisto, 2023-12-27:
The issue still persists on
app.grapheneos.camera 64
TP1A.221005.002.B2.2023122200

empratyush, 2023-12-27:
@hephaisto  on pixel 4 ?

hephaisto, 2023-12-27:
> @hephaisto on pixel 4 ?

yes.

hephaisto, 2023-12-27:
I have to make a few corrections/additions: Now a video is saved. However, recording is stopped without pressing the record-button. This is the sequence of events and the content of the resulting video file:

* Double-press power to switch on camera
* Press record
* Stuff here is recorded normally
* Press power button
  * screen off
*  Stuff here is recorded normally
* ~6s pass
* Video file ends
* more time passes
* Press power button
  * screen on
  * Error message shows

So now at least the video until the power button press is saved, which definitely is an improvement over the previous state. A least the first video is not totally lost. Thanks for that fix so far! However, the video recording still aborts when switching off the screen.

I'm not sure if the OS itself is messing with the camera in this situation. If it is impossible for the camera app to keep access to the hardware, the current behavior is probably a "won't fix" for the camera app, and could only be changed in the OS. If that is the case, I'd accept the current state as "solved" and I'll probably file a report there.

empratyush, 2023-12-28:
This is the expected behaviour camera can only be used while the app is in the foreground.

closed 2023-12-28 by thestinger

hephaisto, 2024-08-14:
@thestinger Yes and no. I reported a state A which was buggy. This was changed to state B, which was not buggy but suboptimal to me. It's okay that it doesn't get changed to C. But B is still an improvement over A.
The original report wasn't about the app being stopped per se, but being terminated in a way which lost all the data.

anonfaded, 2024-08-16:
well you can use other camera apps that are specifically made with a foreground recording feature.

anonfaded, 2024-08-16:
> It's a feature which could be implemented in this app but it's a separate request from what was requested here.

Yeah but he can use other alternatives until its officially added to the main camera app, anyways the chat here is very old and he probably isn't even seeing the latest replies.

PR #384 (2023-10-16): [bugfixes] handle VideoRecordEvent callback correctly.
closes
 #381 
 #347 
 #346 
 #287