GrapheneOS/Camera #382: Camera not working on Pixel 5a after September update

rtonerii, 2023-10-02:
My wife's Google Pixel 5a camera stopped working completely after getting last month's update.  I asked her to wait until this month's release and it "should " be fixed.

Well, it does not function at all.
I tried the Open Camera but it says the camera is not available.

thestinger, 2023-10-03:
This is a hardware issue, not an issue caused by an update.

thestinger, 2023-10-03:
The camera not being available means the hardware isn't working. It's a symptom that's seen after drop damage. It often means that there's a loose connection and it's fairly easily fixable. There's no way an issue like this would have made it past the Alpha and Beta channels. It's not something caused by an update and won't ever be fixed by an update. You need the phone repaired, which can probably be done without returning it for repair by a phone repair shop. Doing it yourself is possible but it's tricky to open up the devices without damaging them.

closed 2023-10-03 by thestinger

rtonerii, 2023-10-09:
Thanks for the reply, I wasn't sure, I just wanted to verify.

rtonerii, 2023-11-12:
Thanks for the update. Whatever you did in last week's update. My wife's cameras (both of them) started working as expected.

thestinger, 2023-11-12:
Nothing.

rtonerii, 2023-11-13:
That's weird, neither of her cameras were working for the last few months and as soon as the update completed and she rebooted both cameras are working great again.

Very weird.