GrapheneOS/Camera #391: EIS not fully working on Pixel 6

Silent-Hunter, 2023-10-26:
EIS works fine when not recording. I can tilt the phone slightly and the image in the viewfinder is stable. But as soon as I start recording, it's as if the EIS is off. It's been like this for over a year.

GrapheneOS build number is UP1A.231005.007.2023102300

Graphene Camera is version 63

soupslurpr, 2023-11-02:
It seems to be intentional

Gets disabled before recording starts here https://github.com/GrapheneOS/Camera/blob/53efbad43aaaeb3a8996c6c77da9d84b2de22dda/app/src/main/java/app/grapheneos/camera/capturer/VideoCapturer.kt#L288

Gets re-enabled when recording stops here
https://github.com/GrapheneOS/Camera/blob/53efbad43aaaeb3a8996c6c77da9d84b2de22dda/app/src/main/java/app/grapheneos/camera/capturer/VideoCapturer.kt#L326

Silent-Hunter, 2023-11-03:
What on Earth? Why? That seems odd.

thestinger, 2023-11-03:
@soupslurpr That code prevents changing the state of the EIS toggle after starting the video. It doesn't disable EIS.

soupslurpr, 2023-11-03:
Ah, sorry for misunderstanding that. Is it a bug then that EIS doesn't seem to work when starting the recording? I noticed this too on my Pixel 7

ASerbinski, 2024-04-22:
I can confirm that reports look valid. Move the EIS switch to "on" and the field of view narrows to provide the feature's required buffer, and shaking the phone around a bit, the image appears to be stabilized. As soon as recording is enabled though, the video no longer appears to be stabilized, but still has the narrowed field of view.

My first thought is that it could be connected to some kind of logic that may disable EIS if the camera has OIS capability. Of note, Pixel 6 does have OIS on the main lens, but if I remember correctly, not on the wide angle or telephoto (pro model) lenses.

MHShetty, 2024-06-17:
Hey everyone,

Could someone please help us out with testing a fix for this issue?

Silent-Hunter, 2024-06-19:
How would one do that?

MHShetty, 2024-06-19:
> How would one do that?

Hey @Silent-Hunter, are you present on Element?

soupslurpr, 2024-06-20:
I can test too :)

Silent-Hunter, 2024-06-20:
> Hey @Silent-Hunter, are you present on Element?
No, sorry.

MHShetty, 2024-06-20:
> > Hey @Silent-Hunter, are you present on Element?
> 
> No, sorry.

Ohh no worries @Silent-Hunter

MHShetty, 2024-06-20:
> I can test too :)

Hey @soupslurpr,

Thanks for volunteering out for helping us with testing a fix for this issue! Unfortunately, I'm unable to share the apk file directly here, so I have pinged you on Element via your ID found on our testing group with the user name somethingdecent/@wowowowowowowowowow:matrix.org

Please accept the chat invite and test the apk for the video stabilization feature on your Pixel 6/Pixel 7 device and let us know the results whenever you're done with the same. Thanks a lot once again!

MHShetty, 2024-06-21:
Hey everyone,

We have tried using the official CameraX API for video stabilization on a Pixel 7 device and that doesn't seem to work with the preview as well currently unfortunately. We have reported this issue on the official CameraX Issue Tracker.

Further updates on this issue can be tracked on https://issuetracker.google.com/issues/346692558 until there's any update directly mentioned on this GitHub issue.

Special thanks to @soupslurpr for helping us out with testing with their Pixel 7!

MHShetty, 2024-08-03:
Hey everyone,

Does EIS work for anyone who was facing issues previously, in the latest release?

soupslurpr, 2024-08-03:
Its working perfectly for me, much smoother with it on vs off, good job!

MHShetty, 2024-08-03:
> Its working perfectly for me, much smoother with it on vs off, good job!

Perfect, glad to know @soupslurpr. Thanks for confirming this!

MHShetty, 2024-08-05:
@Silent-Hunter @ASerbinski Could you please confirm if the issue is fixed on your end in the latest release? We can raise the issue further to the CameraX team just in case if the issue still persists. Thanks a lot for your time and help!

Silent-Hunter, 2024-08-05:
Is 73 the latest release? That's the version I currently have; I'm not sure how to update it beyond that manually, if it's not.

MHShetty, 2024-08-06:
> Is 73 the latest release? That's the version I currently have; I'm not sure how to update it beyond that manually, if it's not.

Yes 73 is the latest version and it should have the changes for EIS included in it

> EDIT: It does seem to be working! It feel less pronounced than when it used to only work in the preview, but it does seem to be doing something.

Ohh glad to know that! Is there any difference between the expected behavior for EIS, and what you're experiencing in the preview and the video that gets recorded?

soupslurpr, 2024-08-06:
I also feel like its less pronounced but its actually a good thing IMO. Before it kind of locked on to a specific spot and didn't allow movement until you moved it a lot and then it again locked on which was annoying but now it allows motion while keeping it smooth. Its a great improvement. The locked mode could be useful I suppose but I definitely think the way it is now makes more sense at least as a default.

Silent-Hunter, 2024-08-06:
What soupslurpr said, basically! It's less pronounced, but the recording now matches the preview, and while the more aggressive mode could definitely be useful for certain things, it's not a necessity.

MHShetty, 2024-08-06:
@Silent-Hunter @soupslurpr Glad to know that this issue has been fixed. Thanks a lot for reporting and confirming this issue!