GrapheneOS/Camera #398: Limited to 12MP on 23MP sensor

Hichen220, 2023-12-04:
Device: Sony Xperia XA2
Sensor Name: Sony IMX300
other CameraX apps can and do shot at a full 23MP resolution, like LineageOS Aperture.

farfromrefug, 2023-12-19:
i am facing the same issue on a a3 2017 and a Sony xperi x2
i have run some tests and Los aperture also have the same issue. same for my app which is using androidx camera.
so I think it is an issue on the lib side.
i have opened an issue  [here](https://issuetracker.google.com/issues/316786569)
hope it is going to get picked up by google

empratyush, 2024-01-13:
Thanks for your bug report. This is something camerax and vendor need to figure out first for now i am going to close this.

closed 2024-01-13 by empratyush

farfromrefug, 2024-01-13:
I had the same issue with the Samsung galaxy a3 2017. and we figured out it was due to a hack of cameraX . maybe it is the same for your device, not sure..
https://issuetracker.google.com/issues/316786569?pli=1