GrapheneOS/Camera #400: Save images and videos in different folders

drugdeliverthatwantprivacy, 2023-12-04:
Save images and videos in different folders.

Currently the camera application saves the images and videos in the same folder (DCIM), I think that most users do not know what DCIM means, it would be good if the main folder of the camera application was called (Graphene Os camera) and inside In this folder there are two folders by default, a folder called (Videos) and another folder called (Photos). This would simplify the transfer of files using applications such as Android Transfer and would make it easier to quickly find items in the file manager.

thestinger, 2023-12-04:
This is the standard location to save them, which we're going to keep as the default. You can set a different location to save the files. It doesn't seem necessary to put them in different locations since they're distinguished by the file extension. Can you explain why you want them in different locations?

drugdeliverthatwantprivacy, 2023-12-04:
The screenshots could be saved within the (downloads) folder in a folder called (Screenshots), the images downloaded from the internet could be saved in another folder within the downloads folder called (images) and the videos downloaded from the internet in a folder inside the downloads folder called (videos)

drugdeliverthatwantprivacy, 2023-12-04:
Because how images and videos are saved today makes those files less accessible to the average user who does not know what DCIM means and it does not seem the most rational to me to save images and videos downloaded from the Internet in the same folder as the images and videos. taken with the camera

thestinger, 2023-12-04:
Screenshots are already saved in Pictures/Screenshots and that has nothing to do with this Camera app.

thestinger, 2023-12-04:
Screenshots are already saved in Pictures/Screenshots and that has nothing to do with this Camera app. Screenshots are not downloads and don't belong there.

thestinger, 2023-12-04:
Images and videos downloaded from the internet are not saved to DCIM/Camera. DCIM/Camera is already a dedicated place for this. It's not clear what the purpose would be of splitting captured images and videos into separate folders.

thestinger, 2023-12-04:
We're going to continue following the standard Android approach to the media directories.

closed 2023-12-04 by thestinger

drugdeliverthatwantprivacy, 2023-12-04:
Even if I don't agree, thank you very much for your quick response :)