GrapheneOS/Camera #403: macro mode toggle for Pixel 8 Pro

thestinger, 2023-12-11:
Not clear if this is feasible to implement and it may require a quite hardware specific API call.

empratyush, 2024-02-15:
Camera2Interop will do the trick afaict.

```
Camera2Interop.Extender(previewBuilder).setCaptureRequestOption(
            CaptureRequest.CONTROL_AF_MODE,
            CaptureRequest.CONTROL_AF_MODE_MACRO
        )
```

azomDev, 2024-02-24:
I want to add that it would be great on the non-pro pixel 8 too if that's possible

io43, 2024-03-11:
yea, very annoying on p8, it is problematic when using the webcam mode too

MHShetty, 2024-08-08:
Hi everyone,

We could try directly implementing this using the Camera2Interop API as @empratyush has suggested, but I have raised a feature request for the same on the CameraX library to have better support for device-specific quirks.
(Link to the filed issue to stay updated: https://issuetracker.google.com/issues/358257249)

lepras, 2025-11-07:
Does this mean we already have macro mode, and can enable it in camera settings somehow, because I can't find a toggle smh.