GrapheneOS/Camera #410: Camera auto-switching is too aggressive (8 Pro)

0xunusedusername, 2024-01-17:
I'm at 1x zoom, intending to use the primary camera at close focus to blur the background, but the camera keeps switching to the ultrawide instead.

The Google Camera app does not behave like this. It would be nice if we could have some lens selection shortcuts or a toggle to disable auto-switching without crossing the zoom threshold.

0xunusedusername, 2024-01-17:
It doesn't though. I'm using them back-to-back and GCam doesn't switch.

> It behaves exactly the same way as Google Camera, switching at the same level it would switch. There's just not a button for going back to exactly 1x. This is a duplicate of #87.

thestinger, 2024-01-17:
It sounds like you're talking about the macro feature, which has a toggle in Google Camera. That's not the same thing.

thestinger, 2024-01-17:
This is a duplicate of https://github.com/GrapheneOS/Camera/issues/403 if it's about adding a toggle for the macro feature.

0xunusedusername, 2024-01-17:
I am not, just talking about a short focus distance (Macro mode actually shows up as 'Not Available' in the quick settings).

> It sounds like you're talking about the macro feature, which has a toggle in Google Camera. That's not the same thing.

thestinger, 2024-01-17:
This zoom-based camera switching is not implemented inside either Google Camera or GrapheneOS Camera. It's done inside the OS, and it's the same implementation. There's limited configuration for it. We don't do anything special. The only part that's implemented in our app is the zoom slider and zoom gestures. The zoom itself is done via a standard Camera2 APi which has camera switching based on zoom level and other factors such as distance to options, light level, etc. depending on the setup.

0xunusedusername, 2024-01-17:
Yeah, I messed up. Google lets you do manual lens selection on the 8p, which got enabled somehow. Would be nice to see it ported, though.

Sorry, my fault.

> This zoom-based camera switching is not implemented inside either Google Camera or GrapheneOS Camera. It's done inside the OS, and it's the same implementation. There's limited configuration for it. [...]