GrapheneOS/Camera #460: Previews are not visible

Jennifer2005x, 2024-08-03:
I have a phone with Android 10 and I love this camera and how private it is, I only have this problem, the previews are seen from the small button but when I want to see them it does not appear, I hope they fix it, it is a good camera

[image]
[image]

Jennifer2005x, 2024-08-03:
> Which version of the camera app are you using?

The last one, the 73

MHShetty, 2024-08-03:
Hi @Jennifer2005x,

Thanks a lot for reporting this issue and sorry for the inconvenience you're facing. 

Did this issue exist on your device in v72?

Jennifer2005x, 2024-08-03:
> Hi @Jennifer2005x,
> 
> Thanks a lot for reporting this issue and sorry for the inconvenience you're facing.
> 
> Did this issue exist on your device in v72?

Since I downloaded the app (approximately a few months) I have had that problem, until now I found out that they have github

MHShetty, 2024-08-03:
> Since I downloaded the app (approximately a few months) I have had that problem, until now I found out that they have github

Ah that seems unusual. Could you please share the logcat while trying to access the in-app gallery? If any internal/app level component logs this issue, we could dive in deeper to see what went wrong

Also, by any little chance, does the image show up when the device is rotated in horizontal orientation?

Jennifer2005x, 2024-08-04:
> Ah that seems unusual. Could you please share the logcat while trying to access the in-app gallery? If any internal/app level component logs this issue, we could dive in deeper to see what went [...]

[2024-08-03-22-38-10.txt](https://github.com/user-attachments/files/16485355/2024-08-03-22-38-10.txt)

And no, the image does not appear when changing the orientation

MHShetty, 2024-08-04:
Hi @Jennifer2005x,

I don't think there's anything relevant in the logs unfortunately. Not being able to reproduce this behavior makes it harder to know what could have gone wrong there, I think we'll need to share a debug build with some more log messages to debug this further. I'll share a debug apk here soon.

Also, are you able to see the details of the photo while clicking on the info icon in the top navigation bar? We could know if it's a rendering problem an issue with loading the saved image uris that way to further narrow down to the exact issue

Jennifer2005x, 2024-08-04:
> Hi @Jennifer2005x,
>
> I don't think there's anything relevant in the logs unfortunately. Not being able to reproduce this behavior makes it harder to know what could have gone wrong there, I think [...]

[image]

MHShetty, 2024-08-04:
The issue seems to be with the rendering in that case. Could be related to the bitmap (image) being too large for the device to render due to memory constraints or other reasons

MHShetty, 2024-08-04:
Could you please try out the debug apk in the below link for this issue? 

https://drive.proton.me/urls/ZE8YSCMNSG#bJjXGQs8RUzU

Jennifer2005x, 2024-08-04:
> Could you please try out the debug apk in the below link for this issue?
> 
> https://drive.proton.me/urls/ZE8YSCMNSG#bJjXGQs8RUzU

[image]
[image]

MHShetty, 2024-08-05:
Hi @Jennifer2005x,

Sorry to keep you waiting. I have maximized our existing image downscaling logic to a high extent to confirm that this is a rendering/memory related issue in the debug apk uploaded in the link below:

https://drive.proton.me/urls/4V5PXPA2R8#OEzrJI1XgZz7

Could you please try this debug apk to check if you're able to view a low-quality image of a recently taken photo? We could look for better image down scaling algorithm or constant to see if that could resolve the issue

Jennifer2005x, 2024-08-06:
> Hi @Jennifer2005x,
>
> Sorry to keep you waiting. I have maximized our existing image downscaling logic to a high extent to confirm that this is a rendering/memory related issue in the debug apk [...]

Funciona, pero, se ve borrosa

[image]

[image]

MHShetty, 2024-08-07:
Hi @Jennifer2005x,

Could you please share the device model you're using for the camera app?

MHShetty, 2024-08-07:
Also, could you please try this test apk on your device? https://drive.proton.me/urls/3YKPK5YC04#YJ4JHLjq2ydu

Jennifer2005x, 2024-08-07:
> Hi @Jennifer2005x,
> 
> Could you please share the device model you're using for the camera app?

It's a Samsung A022M

Jennifer2005x, 2024-08-07:
> Also, could you please try this test apk on your device? https://drive.proton.me/urls/3YKPK5YC04#YJ4JHLjq2ydu

Same

[image]
[image]

MHShetty, 2024-08-13:
Hi @Jennifer2005x,

Could you please test the following debug apk on your device to check if it loads the preview is visible?

https://drive.proton.me/urls/WAJBQ22PJC#cve6AwLBGj7z

Jennifer2005x, 2024-08-13:
It works more or less
[image]
[image]

MHShetty, 2024-08-13:
Glad to know that it's working as expected.

Will make a PR to include this fix soon after a issue with the new implementation gets addressed and after some testing.

Thanks for confirming @Jennifer2005x!

MHShetty, 2024-08-15:
Hi @Jennifer2005x,

Could you please test the in-app gallery of the camera debug build present in the following apk?

https://drive.proton.me/urls/0HAM7AYBEW#VoqTarq9Ypho

Jennifer2005x, 2024-08-15:
It looks good now
[image]
[image]